cta-observatory/ctapipe #3: Test datasets setup

cdeil, 2015-06-24:
As discussed today, I plan to set up a test dataset repo and way to use the test data the way the Sherpa devs did:

* https://github.com/sherpa/sherpa-test-data
* https://github.com/sherpa/sherpa

@kosack - Maybe you could look over the details and their reasoning to set it up exactly like this in https://github.com/sherpa/sherpa/pull/31? If no, that's OK ... I'll just try it out tomorrow and we'll see if we like it.

cdeil, 2015-06-26:
This is mostly done (and documented in the "Getting started" and "Datasets" section of the docs.

TODO:
- [ ] Define decorator to declare optional dependencies and data files needed for tests, so that they are skipped if not present
- [ ] Do we need / want `ctapipe.datasets` with functions to fetch datasets and generate example datasets  (e.g. for benchmarking)? Or should we have such functions elsewhere?

kosack, 2017-05-18:
this is mostly solved by the ctapipe_resources module which is now required for testing

closed 2017-05-18 by kosack