cta-observatory/ctapipe #5: storage of required data

kosack, 2015-06-25:
There are some data items that should be stored along with the repository, for example files containing camera geometries and array layouts.  In the future those would be read directly from the raw data files, but there should be "default" versions included to make it possible to debug code without loading raw data.   Where should this go?  The format and naming wil. be submitted as a redmine issue for the MODEL group.

perhaps to start with:

```
  data/ 
     camera_geometry/
         hess_v0.0_camerageom.fits.gz
         chec_v0.0_camerageom.fits.gz
     array_geometry/
         hess_v0.0_arraygeom.fits.gz
     ...
```

cdeil, 2015-06-25:
@kosack – Can you come by my office?
I'm setting this up at the moment and also have some questions.

cdeil, 2015-06-25:
@kosack – My suggestion would be to keep data files outside the code repo and put them here:
https://github.com/cta-observatory/ctapipe-extra/tree/master/datasets

Then we'll make it easy for devs / users to get at those files.
I've started this, but it's a bit tricky in detail:
https://github.com/cta-observatory/ctapipe/blob/master/ctapipe/utils/datasets.py
https://github.com/cta-observatory/ctapipe/blob/master/docs/datasets/index.rst

Here's a working example though how to access files from the `ctapipe-extra` repo from `ctapipe` tests or examples or docs:
https://github.com/cta-observatory/ctapipe/blob/master/ctapipe/reco/tests/test_hillas.py#L10

OK?

cdeil, 2015-07-02:
@kosack – Close this issue? 

Basically the answer is that you can put up to a GigaByte of example files in any folder structure you like in `ctapipe-extra`, although it would be best to keep that <~ 100 MB and add functions to fetch larger datasets from servers if needed, or generate them by faking data or duplicating small datasets.

closed 2015-07-16 by kosack