cta-observatory/ctapipe #6: astropy-helpers issue

cdeil, 2015-06-26:
@astrofrog or @embray – Could you help me with an issue with the `astropy-helpers` setup in this package?

I first bundled `astropy_helpers` because I didn't want the submodule. Then I always got some warning from `python setup.py` that the git submodule wasn't there. So I changed it and now I have `astropy-helpers` as a git submodule.

But there must be something wrong, I'm getting this error from the sphinx build on travis-ci (which I don't see locally):
https://travis-ci.org/cta-observatory/ctapipe/jobs/68434189#L942
```
Traceback (most recent call last):
  File "setup.py", line 129, in <module>
    **package_info
  File "/home/travis/miniconda/envs/test/lib/python3.4/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/home/travis/miniconda/envs/test/lib/python3.4/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/home/travis/miniconda/envs/test/lib/python3.4/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/home/travis/build/cta-observatory/ctapipe/.eggs/astropy_helpers-1.0.2-py3.4.egg/astropy_helpers/commands/build_sphinx.py", line 120, in run
    ah_path = os.path.abspath(ah_importer.path)
AttributeError: 'NoneType' object has no attribute 'path'
```

And locally there's this:
```
$ python setup.py build_sphinx
git submodule command failed unexpectedly:
error: pathspec 'astropy_helpers' did not match any file(s) known to git.
Did you forget to 'git add'?
```
```
$ git status
On branch master
Your branch is up-to-date with 'origin/master'.
Untracked files:
  (use "git add <file>..." to include in what will be committed)

	astropy_helpers/

nothing added to commit but untracked files present (use "git add" to track)
```
What do I have to do to get the `astropy-helpers` setup corrected in this package?

astrofrog, 2015-06-26:
@cdeil - how did you add the submodule? What do you get for ``git submodule``?

cdeil, 2015-06-26:
@astrofrog – I did a series of commands and screwed up (first calling it `astropy-helpers`, then `astropy_helpers` and manually editing `.gitmodules`. I now have:
```
$ git submodule
 e0c632b7a1b17ffa02872be24af3c98108f13382 ctapipe-extra (heads/master)
$ cat .gitmodules 
[submodule "ctapipe-extra"]
	path = ctapipe-extra
	url = https://github.com/cta-observatory/ctapipe-extra.git
[submodule "astropy_helpers"]
	path = astropy_helpers
	url = https://github.com/astropy/astropy-helpers.git
```

@astrofrog – If you could have a quick look that would be great!
```
git clone https://github.com/cta-observatory/ctapipe.git
```

astrofrog, 2015-06-26:
@cdeil - the submodule isn't currently in the repository. I don't think editing ``.gitmodules`` is enough. Remove the entry from ``.gitmodules`` then do:

```
git submodule add https://github.com/astropy/astropy-helpers.git astropy_helpers
```

Then commit the change. This works for me.

cdeil, 2015-06-26:
@astrofrog – Your suggestions worked. Thanks!

Fixed in 3eac7be85c77dc05d311ed7637bf1c2fcec01799

closed 2015-06-26 by cdeil

embray, 2015-06-30:
I'm a little more curious about the original issue...  You don't need to have astropy_helpers as a submodule to use it.  I wonder if that's a weird issue with the package-template.