cta-observatory/ctapipe #7: How to use hessioxxx from ctapipe?

cdeil, 2015-06-26:
@kosack - How should we use hessioxxx from ctapipe?

I guess the versions are decoupled, so no need to have hessioxxx as a git submodule here.
Could still be convenient though for devs to get it via `make init` (I added a Makefile) and know the location of `hessioxxx` so that we can call `hessioxxx cmake` from `ctapipe setup.py`?

Add the Python wrappers in `hessioxxx` or `ctapipe`?

Should we try to build a conda package for `hessioxxx` and use it as a library, or should I build it from source on travis-ci in the `ctapipe` tests from now?

cdeil, 2015-07-10:
@jacquemier did this: https://github.com/jacquemier/pyhessioxxx

@kosack – What's the conclusion of the email discussions?
What goes in which repo? How should the build be implemented? Can you implement it or should Jean or I do it?

kosack, 2015-09-08:
IT does seem to be a problematic to do the checkout/compile manually, as seen in issue #22 (particularly since hessioxxx is sometimes updated).  I see two ways to proceed:

1. make a conda package for pyhessioxxx, and we can just make it a dependency of ctapipe that can be installed via `conda install pyhessio=version` (that requires making a package release of pyhessioxxx every so often after it's updated)
2. make pyhessioxxx a sub-repo of ctapipe, and thus each commit of ctapipe is tied to a specific commit of pyhessioxxx, similar to ctapipe-extras

I kind of like option 1 better, since having a bunch of sub-repos may start to get complicated, and other dependencies are done via conda package.  but 2 may be easier to implement...

What do you think @jacquemier?

jacquemier, 2015-09-15:
I agree, I will make a conda package for pyhessioxxx.

kosack, 2015-09-15:
Thanks! I think you can just create an appropriate setup.py file (you also need an __init__.py to make it a proper package), and you can automatically use the`python setup.py bdist_conda` to create the package.

jacquemier, 2015-09-16:
It seems to be a bit more complicated.
Do you have submit ctapipe to pypi before created a conda package ?
I try to write a setup.py starting from the ctapipe one.
@cdeil  Why is there some many references to astropy in ctapipe/setup.py and do I need to keep them in pyhessioxxx ?

cdeil, 2015-09-16:
@jacquemier – The mention of Astropy in setup.py is because it's based on the setup.py from the Astropy affiliated package template:
https://github.com/astropy/package-template

Actually the info is pulled in from setup.cfg, and that looks OK:
https://github.com/cta-observatory/ctapipe/blob/master/setup.cfg#L18

Of course, if something else needs to be changed or added to setup.py or setup.cfg, go ahead.

IMO it's really important that all repos we use are moved to the cta-observatory organisation to make it possible / easy to control access.

@kosack – Do you have access already to https://pypi.python.org/pypi/ctapipe ? If no, what's user PyPI username so that I can give you access.

It would be best if PyPI source releases and conda binary releases are done in sync and by one person.
I don't have time. @kosack - do you?

kosack, 2015-09-24:
My username is also 'kosack' on pypy, so you can give me access if you like

cdeil, 2015-09-24:
I've set `kosack` as co-owner for `ctapipe` on PyPI just now.

jacquemier, 2015-10-13:
I'm trying to create an anaconda package of pyhessioxxx
As you know pyhessioxx contains hesioxxx C library.
Can I configure anaconda or astropy_helpers to automatically build a shared library from hesioxxx source file when user will install pyhessioxxx via conda command ?

kosack, 2015-10-13:
I think that is a pretty normal thing to do (many python packages contain some C/C++ extensions), and all you need is to set up `setup.py` correctly. This link shows https://docs.python.org/3.4/extending/building.html  some examples.  Since the build system for hessioxxx is already based on a pretty complex makefile, it may be simplest just to have setup.py run "make" in the hessioxxx directory first, and then use it's more fancy build options to build pyhessio.c.

I don't think you need astropy_helpers at all, in fact, so you can probably get away with a more simple package template if you like.

jacquemier, 2015-10-14:
python setup.py install will produce two shared libraries:
1/ libhessio.so
    - produced by hessioxxx/Makefile
        (with os.system("make CDEBUGFLAGS=\"-g -O2\" DEFINES=\"-DCTA -DCTA_MAX_SC\"")
    - moved to hessioxxx/lib directory

2/ pyhessio.cpython-34m.so 
    - produced by distutils.core.Extension thanks to setup method and its ext_modules
    - moved to $HOME/anaconda3/lib/python3.4/site-packages/

Questions:
1/ After having run python setup.py install command, user need to set LD_LIBRARY_PATH for libhession.so.
Do you know how to avoid that ? For example is there a way to automaticaly copy it to $HOME/anaconda3/lib/python3.4/site-packages ?

2/ hessio.py need to load pyhessio library thanks to numpy
np.ctypeslib.load_library('pyhessio.cpython-34m.so', _path)
How can I resolve _path for user's anaconda library path ?

jacquemier, 2015-10-14:
I have a firswt lead.
When using anaconda, python sys.prefix return $ANACONDA_INSTALL_PATH/anaconda3

kosack, 2015-10-14:
I think we should avoid requireing `LD_LIBRARY_PATH` to be set (especially since it is not the same on MacOS, where it is `DYLD_LIBRARY_PATH` and can be blocked by the OS for security issues). 

the top answer to this question: http://stackoverflow.com/questions/9795793/shared-library-dependencies-with-distutils seems to have a solution, using relative paths.

MaxNoe, 2015-11-07:
You should try to build the libhessio with `conda build` and make also a conda package out of that.
You can take these as orientation:
https://github.com/conda/conda-recipes

Then the libhessio is build with the anaconda libraries and put into the anaconda `lib` folder.

MaxNoe, 2015-11-10:
I succeded building a conda package from the `cta-observatory/hessioxxx`. I had to modify the `src/CMakeLists.txt`.

My `meta.yaml`:
```
package:
  name: hessioxxx
  version: 0.1

source:
  git_url: https://github.com/cta-observatory/hessioxxx
  patches:
    - cmake.patch

build:
  number: 1
  binary_relocation: True
  detect_binary_files_with_prefix: True
```

This is my `build.sh`:
```
#!/bin/sh

cmake . -DCMAKE_INSTALL_PREFIX=$PREFIX
make -j$(getconf _NPROCESSORS_ONLN)
make install
```

And this is the `cmake.patch` file:
```
diff --git src/CMakeLists.txt src/CMakeLists.txt
--- src/CMakeLists.txt
+++ src/CMakeLists.txt
@@ -1,3 +1,6 @@
+set(CMAKE_INSTALL_RPATH "${CMAKE_INSTALL_PREFIX}/lib")
+set(CMAKE_INSTALL_RPATH_USE_LINK_PATH TRUE)
+
 set( HESSIO_SOURCES
                     atmprof.c
                     current.c
@@ -40,6 +43,7 @@ add_library( hessio++ SHARED
              ${HESSIO_SOURCES} ${HESSIO_INCLUDES} )
 
 
+
 # C Executables
 
 add_executable( testio testio.c )
@@ -87,3 +91,7 @@ target_link_libraries( statio hessio++ )
 add_executable( filterio filterio.cc )
 target_link_libraries( filterio hessio++ )
 
+install (TARGETS hessio hessio++ testio read_hess read_hess_nr split_hessio check_trgmask gen_trgmask extract_hess merge_simtel list_histograms add_histograms fcat  testio++ statio filterio
+	RUNTIME DESTINATION bin
+	LIBRARY DESTINATION lib
+)
```

jisacs, 2015-11-10:
Thank you Max,

Once conda package is built and installed, how can I specify the directory containing libhessio.so library ? (/home/jacquem/anaconda3/pkgs/hessioxxx-0.1-1/lib/libhessio.so)
this path is needed by numpy.ctypeslib.load_library in hessio.py

lib = np.ctypeslib.load_library('libhessio.so')

MaxNoe, 2015-11-12:
I added also the python library to the conda recipe in https://github.com/MaxNoe/hessioxxx/tree/pymodule.

The trick is to use the environtment variables `CPATH` and `LIBRARY_PATH` to include the corrrect headers and link `dynamically` to the correct librarys. This does eliminate the need to set the `LD_LIBRARY_PATH`.

kosack, 2015-11-12:
That's great!  Can we now merge all of these developments into one official version in the cta-observatory organization?   What is the new procedure for building and installing hessio and the python wrapper?   Can that be documented somewhere?

cdeil, 2015-11-12:
> Can we now merge all of these developments into one official version in the cta-observatory organization? What is the new procedure for building and installing hessio and the python wrapper? Can that be documented somewhere?

Completely agree!
If someone adds one or several travis-ci builds for `ctapipe` that actually install `hessio` as described in the docs and runs an example or test to show that it works, that would be wonderful.

jacquemier, 2015-11-12:
I built a new package pyhessio thanks to what max produced (thank you Max for your great help). I also removed some hessioxxx unused files
https://github.com/jacquemier/pyhessio

Installation procedure is:
*****************************
git clone https://github.com/jacquemier/pyhessio
conda build pyhessio
conda install --use-local pyhessio
*****************************
I think I cannot create an official version in the cta-observatory organization because I do not have right to do it.

jacquemier, 2015-11-13:
I renamed hessio module to pyhessio to avoid any confusion with ctapipe.io.hessio module.
But feel to restore original name if you prefer.

kosack, 2015-11-13:
I think keeping it pyhessio is fine (to avoid confusing with the original hessio package, which is c-only).  Eventually we can merge the changes into the official hessio repository if Konrad agrees, but right now that has some problems.

By the way, the install procedure doesn't work for me (the shared library is not found when I import it). I'll create an issue for that with details.

MaxNoe, 2015-11-14:
While many _packages_ start with `py` the modules usually do not contain it. E. g. `pyzmq` vs. `import zmq`.

jacquemier, 2016-03-04:
I added a unit test to test ctapipe.io.hessio module, and therefore pyhessio package installation.
I added pyhessio package installation in travis procedure.

closed 2016-03-04 by jacquemier