cta-observatory/ctapipe #8: Release 0.1

cdeil, 2015-06-26:
@kosack – We should release early and often.

So I've added a 0.1 milestone for Friday next week (July 3).

This is a test release where I want to see if making the source tarball works and things like accessing the right version of cta-extra data from a stable release. It will not be advertised to anyone.

IMO it's not important to have any features ready by then, although having direction reco working by the end of next week would be nice.

kosack, 2015-06-29:
That sounds good to me. I will try to also add all of my example code somewhere, and clean up some of the docs.  If I have time, I'll also look into and example of packaging+interfacing C code.

cdeil, 2015-07-02:
@kosack – Sorry I didn't have time to work on `ctapipe` at all this week.

I still think it's useful to make a release for checking / practice.

Do you want to do it or should I? (if there's no issues it only takes ~ 1 hour to do it and some basic checking, but usually with the first release there's some things, like missing files because of the MANIFEST file or whatever ...)
Here's some notes I have for Gammapy how to make a release: https://gammapy.readthedocs.org/en/latest/development/index.html#make-a-gammapy-release

cdeil, 2015-11-04:
@kosack – Any chance you can make a 0.1 release before the PyGamma15 workshop for ctapipe and maybe even make `pip install ctapipe` and / or `conda install ctapipe` work?

kosack, 2015-11-04:
I can probably do that, but the project may soon move to being semi-private (due to the Project Office decisions today), so that may cause me some problems. I'll try to at least make a release of what we have. Ideally I'd like it on a separate conda channel

MaxNoe, 2015-11-05:
Are you saying, that CTA just decided that the analysis software for CTA will be closed source?

kosack, 2015-11-06:
No, It will almost certainly still be open-source (I will strongly push for that), but the development and distribution may be centralized (e.g. we may be required to abandon github + pypi, etc, and use a system installed at the PO like GitLab + a specialized conda channel).  Therefore a simple "pip install" may not work if we can't upload things to the public pypi.  It would still be fairly open, though it's not clear if they want us to start with a private repo or not. It all depends on what is drafted in the consortium agreement by lawyers.

cdeil, 2017-04-22:
Closing this old issue now. Looks like there have been releases (or at least tagged versions) of ctapipe in the past.

closed 2017-04-22 by cdeil