cta-observatory/ctapipe #10: Add missing file: hess_camgeom.fits.gz

cdeil, 2015-06-29:
@kosack – Tests are failing at the moment:
https://gist.github.com/cdeil/6480edb512002e449317#file-gistfile1-txt-L579

Can you please add the missing file to the `ctapipe-extra` repo?

kosack, 2015-06-29:
I can't seem to push to the ctapipe-extra repo. I added several files to my own copy, but doing a git push fails from within the module. It seems to not know where the remote is, so perhaps the submodule clone was not done properly?

cdeil, 2015-06-29:
I'm not terribly familiar with git submodules. You could try doing a fresh `git clone --recursive` of `ctapipe` to make sure it's OK and then try adding the file again. After a quick look at [here](https://git-scm.com/book/en/v2/Git-Tools-Submodules) and [here](http://git-scm.com/docs/git-submodule) I think though that it might be easier to just to a separate clone of `ctapipe-extra` and add the files there, then only move the submodule pointer. That's how I've done it so far. I'll try directly adding things in a test repo with a submodule tomorrow.

kosack, 2015-06-30:
Ok, checking ctapipe-extra out separately worked. The files should be there now

cdeil, 2015-06-30:
@kosack – You still have to update the git submodule pointer in the `ctapipe` repo.
It's still pointing to https://github.com/cta-observatory/ctapipe-extra/commit/e0c632b7a1b17ffa02872be24af3c98108f13382 and now should point to https://github.com/cta-observatory/ctapipe-extra/commit/2b69c31c9a98ca92df30b36ceaed24ce1577e0dc

kosack, 2015-07-01:
I'm not really familiar with how submodules work. How do I update the pointer?

kosack, 2015-07-01:
doing `git submodule ctapipe-extra pull origin master` seems to have worked.

I guess we have to do that each time it is updated, and the check it back in?

cdeil, 2015-07-01:
First look up the SHA of the correct `ctapipe-extra` version. Usually it will be the last commit (e.g. here: https://github.com/cta-observatory/ctapipe-extra/commits/master), at the moment it's https://github.com/cta-observatory/ctapipe-extra/commit/fda08f3ed016469bedb599cd498bf475b9f05365) 

These commands should work:
```
cd ctapipe/ctapipe-extra # go into the git submodule
git pull
git checkout fda08f
cd .. # go into the parent git module
git add ctapipe-extra
git commit -m 'Update ctapipe-extra submodule pointer'
git push origin master # let other's know
```

cdeil, 2015-07-02:
The file is there now and the tests are passing on travis-ci.
https://github.com/cta-observatory/ctapipe-extra/tree/master/datasets

Closing this issue.

closed 2015-07-02 by cdeil