cta-observatory/ctapipe #11: add example of a c-based extension

kosack, 2015-06-30:
See perhaps here on how to  add an C-file + swig wrapper using setup.py's Extension module:
    https://scipy-lectures.github.io/advanced/interfacing_with_c/interfacing_with_c.html

Make example of an algorithm implemented in c as an extension.  This is not quite the same thing as including a C library in the package (like libeventio), which should perhaps be done via a conda package or similar...

cdeil, 2015-06-30:
:+1: to add a swig-wrapped C file as an example.
We should also add a Cython-wrapped one, which is what most scientific Python packages use.

Maybe it would be best to have a `ctapipe.examples` sub-module or something like that, which contains very simple commented examples of interfacing with C code or using Numba?

dneise, 2018-12-11:
Maybe the cython stuff from @MaxNoe in https://github.com/fact-project/pyeventio could serve as a nice example on how to speed up Python with Cython.

MaxNoe, 2018-12-19:
I don't think we should add examples that are not needed here.

As soon as we decided we need an extension for some task, we can add the extension, but I would not add one just for the sake of having an example.

closed 2019-01-03 by MaxNoe