cta-observatory/ctapipe #12: building conda package doesn't see scipy as a dependency

kosack, 2015-07-02:
when running `python setup.py bdist_conda` (or now `make dist`) the conda package is built and installed in a virtualenv.  It all works until the tests are run, and they fail because `scipy` is not found (because it is not instelled in the virtualenv).

setup.py seems to know the following dependencies:

```
    astropy:    1.0.3-np19py34_0 
    ctapipe:    0.0.dev110-py34_0
    numpy:      1.9.2-py34_0     
    openssl:    1.0.1k-1         
    pip:        7.0.3-py34_0     
    python:     3.4.3-0          
    readline:   6.2-2            
    setuptools: 17.1.1-py34_0    
    sqlite:     3.8.4.1-1        
    tk:         8.5.18-0         
    xz:         5.0.5-0          
    zlib:       1.2.8-0    
```

but it's not obvious to me where it gets this list. How can we add scipy to the list of required dependencies? Where are the deps listed?

cdeil, 2015-07-02:
Python has this worst ever way of declaring dependencies and everything else about a package in a Python script, `setup.py`.

There's the `install_requires` that lists `astropy` ... probably all the other ones are `astropy` dependencies:
https://github.com/cta-observatory/ctapipe/blob/master/setup.py#L107
Adding `scipy` there should solve the issue?

Here's an example for `extras_require`:
https://github.com/gammapy/gammapy/blob/master/setup.py#L138

There's also the `pip` `requirements.txt` files ... I don't know if `conda` looks at this:
https://pip.pypa.io/en/latest/reference/pip_install.html#requirements-file-format

Here's a nice setup where the duplication is avoided ... `setup.py` reads the `requirements.txt` files:
https://github.com/rootpy/rootpy/blob/master/setup.py#L120

cdeil, 2015-07-06:
If I put this in `setup.py` it works:
```python
install_requires=['astropy', 'scipy', 'matplotlib']
```

kosack, 2015-07-10:
thanks! somehow I missed that line. I'll make the change

kosack, 2017-05-18:
no longer a pro

closed 2017-05-18 by kosack