cta-observatory/ctapipe #13: Add shower image browser

cdeil, 2015-07-06:
It would be nice to have a shower image browser (with buttons to select the event) for files like https://github.com/cta-observatory/ctapipe-extra/blob/master/datasets/televentlist_img_small.fits.gz .

@kosack – Do you have something like that? If no, would it be best to implement it with matplotlib or QT or IPython widgets or as a small web app (e.g. using bottle)?

kosack, 2015-07-07:
There's only an older demo using cairo that does that so far (and only works with Py27 since I used wxwidgets, which so far is not supported).  However, now that there is a CameraDisplay in the visualization module, it should be trivial to write some code that steps through events.  I need to implement the data reader (already in one of the examples/brainstorm/misc_demos/), and it should work.

If we want to step through events, I can just add an matplotlib event handler (e.g. to trap mouse-clicks to go to next image) for now, and later we can make a nicer GUI if we choose a suitable widget library, or use a web interface.

MaxNoe, 2015-11-05:
I would like to contribute on this. I wrote somthing that goes into this direction (very basic)
for FACT using tkinter: 

https://github.com/MaxNoe/pyfact/blob/master/fact/plotting/viewer.py

kosack, 2015-11-06:
Great! It would be very nice to have a gui where we can step through events and inspect them.  It's not trivial when you have hundreds of telescopes though. I'm in favor of something simple like a overview array display where you can click a telescope and see it's image, and then click a pixel and see the trace, pedetal, etc.  You could also overlay e.g. the shower direction on the array display, etc to get an idea of the event reconstruction, or other parameters.    We don't have to be super fancy, since there is an independent team (working with Inria and others) responsible for developing the "master GUIs" used by operators, and I suspect we will later be able to use what they develop, but that will take a lot of time and we need a GUI now to help with visualizing and debugging algorithms.

MaxNoe, 2015-11-07:
I started playing around with PyQt4 (sadly PyQt5 is not yet supported by anaconda). 

Would be happy for some feedback: https://github.com/MaxNoe/ctapipe/tree/maxnoe/gui

kosack, 2015-11-30:
can you give an example of how to start the gui? maybe add a command-line tool in ctapipe/tools?

MaxNoe, 2015-11-30:
I added an entry point `ctapipe-gui` to start the browser. Or python -m ctapipe.visualisation.pyqtgui`. Add the moment it only displays mock shower images.

kosack, 2015-12-02:
Looks promising so far! Once it is working well, it can be wrapped into a main window with panes or tabs or something (it's actually not trivial to display so much information).  It's already something that is very useful for people developing camera hardware.  Some features that would be nice:

* over-plot multiple pixel-traces in a single plot (optionally)
* use visualization.ArrayDisplay and  for the array part
* be able to click a position (or select a range) in the trace plot and update the camera plot with the integral over that range, or the single slice

all of that probably requires a more complex GUI window, but should be possible.  The MVC style model should be used to develop such a GUI so one could have multiple "views" of the same data.

I know there is a group working on designing GUIs and displays for ACTL, so it may be nice to look at what they came up with (they worked with Inria and others to come up with a design, which think are just sketches, not implementations)

TarekHC, 2016-03-16:
Brainstorming @ Berlin by Jean, Max, Gernot and Tarek

GUI CTA MC event viewer.

Qt was proposed for the implementation (it is not a new dependency, and has everything we need).

General ideas: 
- [H] Gui should be able to open existing MC files
- [L] It should be able to export any shown image (trivial?)
- [H] "Tabs" would be useful. Better than several "popping" windows.
- [H] Gui could be launched as a tool
- [H] Gui could be also used along the ctapipe analysis module.
- [M] Gui should be able to run and show steps of the analysis (for one or a bunch of events) 
- [L] Leave text boxes to configure analysis parameters.
- [L] Provide methods to select the events for display.
- Fields to introduce any kind of cut
- Terminal showing the log of the analysis
- [L] Create api to data containers to be able to add information for the gui. 

Global array information:
- Info tab on the MC (and real) event metadata
- Array layout: Triggered telescopes (color scale for image size)
- Reconstructed direction (weight, reconstructed core/real core), for now in  ground coordinates.
- Status of telescope. 

Individual telescope info: 
- [H] Whole camera viewer 
- [L] Show camera and pixel status
- [H] Raw camera image
- [M] Cleaned image (with default or custom cleaning levels/algorithms)
- Overlay of some implemented image reconstructions (Hillas, muon rings, image templates, etc...)
- Mean charge vs time slice.
- Add a slider to select the shown time slice (not by default).

Individual pixel information
- Selectable pixels showing: charge vs time slice, position of integrated window, pedestal, etc...

cdeil, 2016-03-16:
Comment from the sidelines: have you considered web GUI instead of QT?
I guess this is a fundamental decision and if you do QT now it'll be hard to change later on without throwing away most GUI code.

Are there important pros and cons for QT or web?
Won't web GUI be better for browsing data that's in a data center?

TarekHC, 2016-03-16:
Hi @cdeil. We will be still discussing about this after the coffee break, so we will also talk about your comments. Please let us know about all your ideas on the topic!! We will post here everything we come up with.

cdeil, 2016-03-16:
I don't have much to add to this discussion web vs desktop GUI.
Naively I would think that the web is the future and better allows for browsing data that's in a data center, no?
But I also know other modern astronomy projects starting now are still choosing QT.

It was just an honest question from my side: which do you choose for now and why?

MaxNoe, 2016-03-16:
@cdeil QT integrates well with the existing matplotlib CameraDisplay and plotting stuff.
A web interface will probably also be needed. 

The thought was mainly to provide the camera guys with a first tool to look into raw data and the people developing algorithms also need visual feedback.

maxnoe, 2020-10-26:
We have `ctapipe-display-dl1` since quite a while now.

closed 2020-10-26 by maxnoe