cta-observatory/ctapipe #14: Shower image example doesn't work

cdeil, 2015-07-06:
@kosack – I've moved this example from a `__main__` section in the `ctapipe.reco` code to the docs
https://github.com/cta-observatory/ctapipe/blob/master/docs/reco/shower_example.py
https://cta-observatory.github.io/ctapipe/reco/index.html#getting-started

I think it's outdated though and doesn't work?
Should it be replaced with one of the examples in the `examples` folder or should it be fixed?

Generally it would be good to have some working examples in the docs and not just in the examples folder.

kosack, 2015-07-07:
I think it's fixed now.  Is it best to put examples in the docs dir or in the examples dir? I was able to include an example in the docs using the following in the sphinx doc (it seems to copy the example correctly, so the docs are still independent) :
<pre><code>
  .. literalinclude:: ../../examples/camera_display.py
   :language: python
</pre></code>

closed 2015-07-07 by kosack

cdeil, 2015-07-07:
Yes, Sphinx copies the example file into the docs build folder if you use a relative path.

I don't have a good strategy for where to put docs, examples, notebooks.
In Gammapy it's a bit of a mess right now: https://gammapy.readthedocs.org/en/latest/tutorials/index.html#python-example-scripts

Maybe keep examples all in the `examples` folder for now because it's easier to find / run for users?