cta-observatory/ctapipe #16: Module structure: where to put ObsConfig and provenance/configuration?

kosack, 2015-07-07:
I'm starting to implement some of the data model group's concept for an ObsConfig, which is a class the contains a hierarchy of the description of the array layout, telescope optics, camera layout, etc.  That is independent of the "configuration service" (which handles the parameters of the algorithms).  I am also looking into an implementation of the _provenance_ data (which is actually quite closely connected to the configuration service, and is maybe the same thing). 

Where should those go?  (right now it's in `ctapipe.io`, but that doesn't really make sense, since that should really just include data readers/writers).

Should we create a module like `ctapipe.core`  or `ctapipe.config`?

I want to avoid 100s of sub-modules , but there should be a somewhat clear hierarchy.

cdeil, 2015-07-07:
Finding a good hierarchy from the start is very hard.
It might be easier to not worry about it much now and just re-organise sub-packages as things get added and it becomes more clear what's what.
I've had to do this a few times for Gammapy and probably will again before 1.0.

That said ... the current [ctapipe.io](https://cta-observatory.github.io/ctapipe/io/index.html) seems to be mostly about camera geometries and that's weird. Yes, you have to read / write camera geometries, but that's pretty much true for most things that you want to be able to read / write them to file and is not an argument to go in `ctapipe.io`.

What you describe above sounds a little bit like `telspec` or `obsconfig` or `arrayconfig` to me ... maybe those would be good sub-packages for now?

kosack, 2015-07-10:
yeah, I think I'll move camera configs and things out of `io`, since that is really for reading "bulk" event data, not instrument configs.

kosack, 2015-12-02:
this has been moved to #47 and #48

closed 2015-12-02 by kosack