cta-observatory/ctapipe #18: Commit example file for read_hessio.py to ctapipe-extra

cdeil, 2015-07-10:
@kosack – Can you please commit this example file to the ctapipe-extra repo?

https://github.com/cta-observatory/ctapipe/blob/master/examples/read_hessio.py#L7
```
filename = "/Users/kosack/Data/CTA/Prod2/proton_20deg_180deg_run32364___cta-prod2_desert-1640m-Aar.simtel.gz"
```

If it's too large (say > 10 MB), do you have a smaller example file?

cc  jacquemier

kosack, 2015-07-16:
Unfortunately the example code is so far 1.4 Gb!   CTA simulations are not small... But it can be downloaded using the CTA password from a central location, so maybe I can add instructions for that (see the Data Model Wiki page on Redmine)

cdeil, 2015-07-16:
Yes, writing instructions or even a script that asks for a password and downloads the files could be useful.

But for testing we still need a much smaller file in the `cta-extra` repo.
I know its painful to waste time preparing example and test data, but it's important.

kosack, 2015-07-16:
Actually, I found that truncating the file to something manageable, it still works (with only a handfull of events).  So now if you update, the example should work (As long as the path to `pyhessioxxx` is in your PYTHONPATH)

closed 2015-07-16 by kosack