cta-observatory/ctapipe #20: change tools in ctapipe/tools to use new Configuration classes

kosack, 2015-08-25:
right now, we only have a few example "tools" (executable programs), that eventually need to have a common interface.  A good place to start is to migrate them to use the configuration class added recently by Jean.  The only tools that are there so far are `ctapipe-info` (used to list what tools are available as well as other version info),  and `ctapipe-camdemo` (an example tool). Both do some command-line handling that can be migrated.

jacquemier, 2015-08-26:
Karl, 

The implementation of the use case "Configure a pipeline" is highly dependent of the pipeline implementation we will choose. 
I think we should start by implementing a simple pipeline first. 

What do you think ? 

https://forge.in2p3.fr/projects/pipelines-use-cases/wiki/ConfigureUseCase 

Cheers, Jean

jacquemier, 2015-08-28:
I have just implemented a simple pipeline to test some configuration implementation. 

Do you want that I share it in ctapipe/example ? 

Cheers, Jean 

----- Mail original -----

De: "Jean Jacquemier" <Jean.Jacquemier@lapp.in2p3.fr> 
À: "cta-observatory/ctapipe" <reply@reply.github.com> 
Envoyé: Mercredi 26 Août 2015 13:11:56 
Objet: Implementation of use case: "Configure a pipeline" 

Karl, 

The implementation of the use case "Configure a pipeline" is highly dependent of the pipeline implementation we will choose. 
I think we should start by implementing a simple pipeline first. 

What do you think ? 

https://forge.in2p3.fr/projects/pipelines-use-cases/wiki/ConfigureUseCase 

Cheers, Jean

kosack, 2015-08-28:
yes that would be great!  And I agree- we need to explore the use cases.  I think there are possibly 2 levels of things that are configured. 

* The executable tools have a  set of user-configurable options
* The algorithms used in a tool themselves have some options, that may be left open to the user  
  - those could be "automatically" added to the list of configuration options at the algorithm level
  - or they could be added at the tool level to the list of tool configuration options (on a tool-by-tool basis)

I'm not sure which is better. In HESS we do the automatic method, but then you get a list of 100s of options for each tool, whereas perhaps a hand-edited list is simpler (but less flexible)

closed 2017-05-18 by kosack