cta-observatory/ctapipe #22: missing hessio python module

moralejo, 2015-09-07:
I don't manage to load the module hessio. I have already included pyhessioxxx in my PYTHONPATH directory, yet I keep getting this message:

>>> from ctapipe.io.hessio import hessio_event_source
the `hessio` python module is required to access MC data: No module named 'hessio'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/nfs/cta-ifae/moralejo/CTA/Pipeline/ctapipe/ctapipe/io/hessio.py", line 19, in <module>
    raise err
  File "/nfs/cta-ifae/moralejo/CTA/Pipeline/ctapipe/ctapipe/io/hessio.py", line 15, in <module>

kosack, 2015-09-08:
if pyhessioxxx is in your python path, you should be able to do `import hessio` (even if ctapipe isn't installed).  Check  that that works first. If not, it's a problem with your PYTHONPATH.  I think you need to have /path/to/pyhessioxxx/  in the path, not the parent directory.

moralejo, 2015-09-08:
On Tue, Sep 8, 2015 at 9:41 AM, Karl Kosack <notifications@github.com>
wrote:

> if pyhessioxxx is in your python path, you should be able to do import
> hessio (even if ctapipe isn't installed). Check that that works first. If
> not, it's a problem with your PYTHONPATH. I think you need to have
> /path/to/pyhessioxxx/ in the path, not the parent directory.
>
May be I do not have the right version of pyhessioxxx. Even with
pyhessioxxx fully in the path I cannot do "import hessio".  I can do
"import hessioxxx", though, but that is not what the example needs.
Specifically, the command

from ctapipe.io.hessio import hessio_event_source

keeps failing at "import hessio"

Where shall I get pyhessioxxx? I just looked for it in github, I could not
find a link in the example pages.

Thanks,

A.

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

-- 
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

kosack, 2015-09-08:
Ah, I think you just need to go one directory lower in your pythonpath, and that should work.  I think the problem is that you have the parent directory of pyhessioxxx in your path, but not pyhessioxxx itself (which contains the hessio.py library file).  We do need to restructure that module to be a correct python module eventually (now it's somehwhat hacked). try making sure the pyhessioxxx/ directory is at the end of the one in your path.  Can you post your PYTHONPATH here?

moralejo, 2015-09-08:
> Ah, I think you just need to go one directory lower in your pythonpath,
> and that should work. I think the problem is that you have the parent
> directory of pyhessioxxx in your path, but not [...]
I have both, i.e.

echo $PYTHONPATH
/nfs/cta-ifae/moralejo/CTA/Pipeline/anaconda3/lib/python3.4/site-packages:/nfs/cta-ifae/moralejo/CTA/Pipeline/anaconda3/lib/python3.4/site-packages/pyhessioxxx

Inside pyhessioxxx I see no hessio.py, just hessioxxx.py. This is why I
think that may be I have the wrong version of pyhessioxxx. Where can I get
the right one? (I think it is not explained in the page).

A.

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

kosack, 2015-09-08:
Try checking out pyhessioxxx again from the github page (or do `git checkout master; git pull` to update it in-place)

kosack, 2015-09-08:
Perhaps in the end we should just include it as a sub-repo, like ctapipe-extra... that way the correct version is automatically checked out.  Another option would be to make it into a correctly-formatted python package (with an __init__.py and setup.py), create a conda package,  and set it as a dependency. Not sure which is a cleaner way to do it. I'll make an issue for it.

cdeil, 2015-09-08:
+10 to have a proper build / install script for everything.
Fiddling with PYTHONPATH will always give issues ...

For building conda packages we possibly could include our packages here:
https://github.com/astropy/conda-builder-affiliated
or clone that setup and build our own.

jacquemier, 2016-03-04:
moralejo, did you managed to install pyhessio properly ?
Could we close this issue ?

moralejo, 2016-03-08:
Hi,

I could not try again until now. I have just installed it (not yet tested),
and it seemed to work.

By the way, there is a small inaccuracy in the instructions in

https://cta-observatory.github.io/ctapipe/getting_started/index.html

conda install pyhessio

should be

conda install --use-local pyhessio

Best regards,

Abelardo

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

moralejo, 2016-03-08:
Tested now with the example scripts, it works. You can close the issue.
Thanks!

A.

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

jacquemier, 2016-03-09:
Thank you for the feed back, 

I am aware of small inaccuracy in the instructions in; I have corrected it, but we have a problem for updating documentation on github. 

Regards, Jean 

| De: "moralejo" <notifications@github.com>
| À: "cta-observatory/ctapipe" <ctapipe@noreply.github.com>
| Cc: "Jean Jacquemier" <jacquem@lapp.in2p3.fr>
| Envoyé: Mardi 8 Mars 2016 18:16:46
| Objet: Re: [ctapipe] missing hessio python module (#22)

| Hi,

| I could not try again until now. I have just installed it (not yet tested),
| and it seemed to work.

| By the way, there is a small inaccuracy in the instructions in

| https://cta-observatory.github.io/ctapipe/getting_started/index.html

| conda install pyhessio

| should be

| conda install --use-local pyhessio

| Best regards,

| Abelardo

|

| > <https://github.com/cta-observatory/ctapipe/issues/22#issuecomment-192223933>
| > .

| --

closed 2016-03-09 by jacquemier