cta-observatory/ctapipe #28: CameraDisplay colorbar breaks when multiple displays per figure

kosack, 2015-11-03:
CameraDisplay now uses `plt.sci(self.pixels)` to set the *current image* to update the color bar if it exists. 

This fails with a ValueError if more than one plot is on a figure (see e.g. `examples/camera_display_multi.py`).  

For now, I catch the ValueError and ignore it, but that is not a good solution as the color bar does not get updated in that case.

MaxNoe, 2015-11-03:
Yes, my bad. 
`plt.sci()` sets the current image on the current ax (`plt.gca()`)

If the provided image is not on the same axes as `plt.gca()` it throws a value error. 
I will a pull request for the fix.

closed 2015-11-04 by kosack

MaxNoe, 2015-11-04:
The matplotlib guys suggest to get rid of it and let the user use the object oriented interface.

Sorry for the inconvenience.

kosack, 2015-11-05:
What does that mean exactly? Is there a correct object-oriented way to keep the color bar updated?

MaxNoe, 2015-11-05:
Sorry for the misunderstaning: `plt.sci()` sets the current image, so that one can call `plt.colorbar()` without explicitly giving the mappable. It does not update the colorbar if the image changes. But this is possible.

commit 86c1dd4c (2015-11-04): Merge pull request #30 from MaxNoe/master

call self.axes._sci instead of plt.sci, fix #28
commit ff950085 (2015-11-04): call self.axes._sci instead of plt.sci, fix #28