cta-observatory/ctapipe #34: Test fail if optional dependency matplotlib not present

cdeil, 2015-11-05:
I think matplotlib should be an optional dependency for ctapipe and `python setup.py test` should succeed if it's not present.

But currently there's a module-level import of matplotlib in `ctapipe/visualization/mpl.py:6` and then pytest errors out trying to collect tests:
https://travis-ci.org/cta-observatory/ctapipe/jobs/89389169#L613

In Astropy / Gammapy this is solved by putting the imports of the optional packages like matplotlib into functions / methods instead of the start of the file, and then by marking tests to declare optional dependencies those tests are skipped if the dependency is not present (see https://gammapy.readthedocs.org/en/latest/api/gammapy.utils.testing.requires_dependency.html).

It should also be possible to configure the pytest test collection to avoid collecting tests from certain modules / packages if a dependency needed there isn't present.

I don't plan to fix this, but merge #33 as-is, so CI will still be broken in master for now.

kosack, 2015-11-05:
That sounds like a reasonable solution. I will try to make those tests optional similar to how you did it in gammapy.

kosack, 2015-11-05:
Hmm. I didn't quite succeed. From the pytest documentation, I see that i can use `pytest.importorskip()` in a test to skip all tests in the module if the import fails. I thought that would work, but since the tests are inside the visualization module, at some point there is a `from visualization import *` that happen implicitly when importing the visualization.tests module, which again imports matplotlib.

I'm looking for a solution.

cdeil, 2015-11-05:
Note that the error occurs during test collection, not test execution.

There should be a way to configure pytest test collection to ignore visualisation if matplotlib isn't present.
But I didn't have to figure that out myself yet.

In Gammapy I have module-level imports of an optional dependency as well, but for some reason the pytest collector doesn't complain about that, i.e. I think pytest doesn't try to find tests there:
https://github.com/gammapy/gammapy/blob/master/gammapy/scripts/catalog_browser/app.py#L2

Let me know if you figure out how this stuff works!

closed 2017-06-20 by kosack