cta-observatory/ctapipe #40: What does n_channels mean?

MaxNoe, 2015-11-10:
Could you explain what the channels mean for the cameras? 
These are  not the pixels, right? Is it lowgain / highgain?

jacquemier, 2015-11-12:
Yes, I confirm,channels mean lowgain/highgain 

----- Mail original -----

kosack, 2015-11-12:
Yes, "channels" generically refer to "data readouts" per pixel (front-end
signal processing + digitization), which in our cameras are associated with
different front-end amplifier gains that are used before digitization,
though in principle they could be used for other purposes. We have multiple
amplifiers to achieve a larger dynamic range in the hardware (except
FlashCam which uses a non-linear amplifier to compress the dynamic range)

closed 2015-11-16 by kosack