cta-observatory/ctapipe #48: Improve instrumental data classes

kosack, 2015-12-02:
__Make  `CameraGeometry` and `ArrayLayout` subclasses of `astropy.table.Table`?__

Right now we have 2 instrumental data classes: `CameraGeometry` and `ArrayLayout` that will eventually be merged into a comprehensive instrument model.  However, their implementation leaves a lot to be desired. 

Currently, `CameraGeometry` is simple a class with members like `pix_x`, `pix_y` that should contain an array of pixel positions.  There is separate support to read them from various formats. 

There is a common use case (working with  cleaned images) that is made difficult by this implementation. After an image is cleaned, one has a boolean _mask_ that can be applied to the image data.  To apply that also to the pixel positions requires masking each member of `CameraGeometry` manually:
```python
geom = CameraGeometry(...)
image = ...
mask = tailcuts_clean(geom, image)

cleaned_image = image[mask]
cleaned_x = CameraGeometry.pix_x[mask]
cleaned_y = CameraGeometry.pix_y[mask]
```
etc.

It would be nicer to allow slicing on the whole `CameraGeometry` object, but currently it contains both vector quantities and scalar quantities like `pix_type`.  

```python
cleaned_image = image[mask]
cleaned_geom = geom[mask]
```

One suggestion: use `astropy.table.Table` as the base class for all instrument classes.  

* Advantages
  * get automatic "slicing" of all data (as above)
  * get automatic read/write to all table formats (so data model is directly connected to the class)
  * get nice print-outs (pretty-printed columns, html in notebook mode)
  * can "browse" the data structures in a web interface
  * could extend to add more reader/writer routines like the current classes
* Disadvantages:
  * can access members by "dot" syntax (would need `geom['pix_x'] rather than `geom.pix_x`)
  * scalar quantities would need to be in the metadata (which is probably correct anyhow), e.g. `geom.meta.pix_type = 'hexagonal'`

If implemented this way, you can do:
```python
geom = CameraGeometry.from_file("test.simtel.gz",5)
geom.write("CT5.fits")
```

cdeil, 2015-12-02:
@kosack - Sounds like it's worth a try.

A word of warning: my impression is that `Table` is a pretty complex (but well written) class, so it can be complex to sub-class and get the exact behaviour you want (e.g. when it comes to the constructor, `read`, `write` and `info` methods.

Concerning the disadvantages you point out: I think there you could quite easily achieve the behaviour you want ... use something like the [bunch pattern](https://github.com/dsc/bunch/blob/master/bunch/__init__.py#L33), to expose columns and meta dict entries directly as top-level class members that can be accessed and discovered via `geom.*`, i.e. `__getitem__`.
If you want this in several classes, making a `NiceTable(Table)` class that implements this and then sub-classing that one might be a good pattern.

We have a few Table sub-classes in Gammapy (e.g. `EventList`) and in other cases decided to use composition, i.e. have a `Table` as a data member (e.g. `SourceCatalog`) ... we're still experimenting and it'll be interesting to see how you use `Table` in `ctapipe`.

kosack, 2015-12-03:
I thought about that as well... it does sounds good to perhaps have a higher-level wrapper to Table (or even using a table as a private internal member, not superclass,  but that also ends up complicated).  

I used something similiar to  the `Bunch` idea in the `ctapipe.core.Container` classes like 'io.RawData', essentially making a class that allows new data to be added, but not automatically (to prevent typos) . Bunches seem to be similar to  `collections.namedtuple` (except those are immutable), and that's what I originally used for `CameraGeometry` (but eventually went to a pure class to implement more functionality).

kosack, 2015-12-03:
as an interim test, I implemented a `CameraGeometry.to_table()` method that allows one to convert between them. I already ran into one problem: I store the pixel neighbors as an  adjacency list (array of linked lists), and amazingly Table has no problem with it, until you try to write it out and then most io backends fail since they don't support var-length arrays.  I can probably solve that by encoding the list differently (maybe a masked array of length max(neighbors) for example, with NaNs in the columns where no neighbor is present)