cta-observatory/ctapipe #54: build_sphinx is the last failing test on travis.

jacquemier, 2016-03-04:
build_sphinx is the last failing test on travis.
It return none-zero value because of warnings, not error.
The easy solution is to remove '-w' option, but detect warning are clear advantages.
Does someone know how to remove this 3 warnings ?

jacquemier, 2016-04-08:
Resoved by #94

jacquemier, 2016-04-08:
On my system there is no more sphinx warning.
But on travis, there are still some warnings.
- warnings came from Matplotlib:
```
"'Matplotlib is building the font cache using fc-list. This may take a moment"
-/home/travis/build/cta-observatory/ctapipe/astropy_helpers/astropy_helpers/sphinx/ext/docscrape.py:120: UserWarning: Unknown section Parameters:
  warn("Unknown section %s" % key)
/home/travis/build/cta-observatory/ctapipe/astropy_helpers/astropy_helpers/sphinx/ext/docscrape.py:120: UserWarning: Unknown section Returns:
  warn("Unknown section %s" % key)
/home/travis/build/cta-observatory/ctapipe/astropy_helpers/astropy_helpers/sphinx/ext/docscrape.py:120: UserWarning: Unknown section Example
  warn("Unknown section %s" % key)
/home/travis/build/cta-observatory/ctapipe/astropy_helpers/astropy_helpers/sphinx/ext/docscrape.py:120: UserWarning: Unknown section Note:
  warn("Unknown section %s" % key)
```

We must have a ctapipe that passes travis to facilitate code review and pull request.
I suggest to remove the "-w" sphinx option until we understand how to  supress these warnings.

Does it sounds good for you ?

MaxNoe, 2016-04-10:
This warning is issued when matplotlib runs for the first time after  installation.
It searches for installed fonts in the system.

closed 2017-06-20 by kosack