cta-observatory/ctapipe #55: How to update ctapipe documenation on https://cta-observatory.github.io

jacquemier, 2016-03-04:
I modified phyhessio installation procedure on ctapipe documents (ctapipe/doc/getting_started/index.rst) but it seems that upload to documantion page is not automatic

kosack, 2016-03-08:
I think you have to type "make doc-publish" in ctapipe (that will upload it).  Originally we had used readthedocs.org to generate it automatically on GIT commmit, but for some reason there were problems with that (perhaps the dependencies)

kosack, 2016-03-08:
I just ran it, in case you don't have permission

jacquemier, 2016-03-08:
I also ran it, So I confirm that I have permission

but documentation does not change on https://cta-observatory.github.io/ctapipe/getting_started/index.html

According to docs/getting_started/index.rst file:
conda install pyhessio 
must be replaced by 
conda install --use-local pyhessio

kosack, 2016-03-09:
It worked for me, but I first did `make clean; make doc; make doc-publish` to be sure everything was built correctly. It takes a few seconds for the new page to appear after you have done that.

closed 2016-03-09 by kosack

jacquemier, 2016-03-16:
It still doesn't work for. 

make clean; make doc; make doc-publish commands finish without error. 
But it does not update the documentation page. 
Do I have to execute these commands on a special branch (gh-pages) ? 

| De: "Karl Kosack" <notifications@github.com>
| À: "cta-observatory/ctapipe" <ctapipe@noreply.github.com>
| Cc: "Jean Jacquemier" <jacquem@lapp.in2p3.fr>
| Envoyé: Mercredi 9 Mars 2016 13:34:38
| Objet: Re: [ctapipe] How to update ctapipe documenation on
| https://cta-observatory.github.io (#55)

| It worked for me, but I first did make clean; make doc; make doc-publish to be
| sure everything was built correctly. It takes a few seconds for the new page to
| appear after you have done that.

| —
|