cta-observatory/ctapipe #65: InstrumentModule read_hessio fails reading prod3 file

MaxNoe, 2016-03-16:
```{python}
from ctapipe.instrument.InstrumentDescription import load_hessio
from telescopes, cameras, optics = load_hessio('/media/Max1TB/gamma_20deg_0deg_run10251___cta-prod3-merged_desert-2150m-Paranal-subarray-3_cone10.simtel.gz')

camera = cameras['CameraTable_VersionFeb2016_TelID9']

```

This results in

```{python}
IndexError                                Traceback (most recent call last)
/home/maxnoe/.local/anaconda3/envs/ctapipe/lib/python3.5/site-packages/IPython/core/formatters.py in __call__(self, obj)
    697                 type_pprinters=self.type_printers,
    698                 deferred_pprinters=self.deferred_printers)
--> 699             printer.pretty(obj)
    700             printer.flush()
    701             return stream.getvalue()

/home/maxnoe/.local/anaconda3/envs/ctapipe/lib/python3.5/site-packages/IPython/lib/pretty.py in pretty(self, obj)
    381                             if callable(meth):
    382                                 return meth(obj, self, cycle)
--> 383             return _default_pprint(obj, self, cycle)
    384         finally:
    385             self.end_group()

/home/maxnoe/.local/anaconda3/envs/ctapipe/lib/python3.5/site-packages/IPython/lib/pretty.py in _default_pprint(obj, p, cycle)
    501     if _safe_getattr(klass, '__repr__', None) not in _baseclass_reprs:
    502         # A user-provided repr. Find newlines and replace them with p.break_()
--> 503         _repr_pprint(obj, p, cycle)
    504         return
    505     p.begin_group(1, '<')

/home/maxnoe/.local/anaconda3/envs/ctapipe/lib/python3.5/site-packages/IPython/lib/pretty.py in _repr_pprint(obj, p, cycle)
    692     """A pprint that just redirects to the normal repr function."""
    693     # Find newlines and replace them with p.break_()
--> 694     output = repr(obj)
    695     for idx,output_line in enumerate(output.splitlines()):
    696         if idx:

/home/maxnoe/.local/anaconda3/envs/ctapipe/lib/python3.5/site-packages/astropy/table/table.py in __repr__(self)
    771 
    772     def __repr__(self):
--> 773         return self._base_repr_(html=False, max_width=None)
    774 
    775     def __unicode__(self):

/home/maxnoe/.local/anaconda3/envs/ctapipe/lib/python3.5/site-packages/astropy/table/table.py in _base_repr_(self, html, descr_vals, max_width, tableid, show_dtype, max_lines, tableclass)
    759             self, tableid=tableid, html=html, max_width=max_width,
    760             show_name=True, show_unit=None, show_dtype=show_dtype,
--> 761             max_lines=max_lines, tableclass=tableclass)
    762 
    763         out = descr + '\n'.join(data_lines)

/home/maxnoe/.local/anaconda3/envs/ctapipe/lib/python3.5/site-packages/astropy/table/pprint.py in _pformat_table(self, table, max_lines, max_width, show_name, show_unit, show_dtype, html, tableid, tableclass, align)
    494             lines, outs = self._pformat_col(col, max_lines, show_name=show_name,
    495                                             show_unit=show_unit, show_dtype=show_dtype,
--> 496                                             align=align_)
    497             if outs['show_length']:
    498                 lines = lines[:-1]

/home/maxnoe/.local/anaconda3/envs/ctapipe/lib/python3.5/site-packages/astropy/table/pprint.py in _pformat_col(self, col, max_lines, show_name, show_unit, show_dtype, show_length, html, align)
    240                                                show_length=show_length,
    241                                                outs=outs)
--> 242         col_strs = list(col_strs_iter)
    243         if len(col_strs) > 0:
    244             col_width = max(len(x) for x in col_strs)

/home/maxnoe/.local/anaconda3/envs/ctapipe/lib/python3.5/site-packages/astropy/table/pprint.py in _pformat_col_iter(self, col, max_lines, show_name, show_unit, outs, show_dtype, show_length)
    405                     else:
    406                         col_str = (format_func(col_format, col[(i,) + multidim0]) +
--> 407                                   ' .. ' +
    408                                   format_func(col_format, col[(i,) + multidim1]))
    409                 else:

astropy/table/_column_mixins.pyx in astropy.table._column_mixins._ColumnGetitemShim.__getitem__ (astropy/table/_column_mixins.c:900)()

astropy/table/_column_mixins.pyx in astropy.table._column_mixins.base_getitem (astropy/table/_column_mixins.c:786)()

astropy/table/_column_mixins.pyx in astropy.table._column_mixins.column_getitem (astropy/table/_column_mixins.c:837)()

IndexError: index 0 is out of bounds for axis 1 with size 0
```

MaxNoe, 2016-03-16:
These are the FlashCam MST and DCSSTs

kosack, 2017-06-19:
the instrument module is being largely reworked, so this issue is no longer valid

closed 2017-06-19 by kosack