cta-observatory/ctapipe #67: Structure of the instrument module

MaxNoe, 2016-03-17:
In my opinion having classes for the instrument description might be a nicer interface.

The different loading and writing functions could then be class methods and be used like this:

```{python}
from ctapipe.instrument import InstrumentDescriptiion

instrument = InstrumentDescription.from_hessio('/path/to/file')
instrument.to_fits('instrument.fits')
```

kosack, 2016-03-18:
agreed. This same concept is mentioned in #49

kosack, 2016-03-18:
Similar to your example, this is from the discussion in #49 :

```python
from instrument import OpticsDescription

od = OpticsDescription.from_file("test.eventio.gz")
od = OpticsDescription.from_metadata(obsid=123456, telid=56)
```

closed 2017-09-27 by kosack