cta-observatory/ctapipe #69: hessio_event_source resuses the same container

MaxNoe, 2016-03-17:
The hessio_event_source reuses the same container again and again, this
has the strange side effect, that you cannot get a list of events from it:

```
from ctapipe.io.hessio import hessio_event_source

events = list(hessio_event_source('path/to/file', max_events=10))
```

This gives you 10 references to the same object, the last read event.

kosack, 2016-03-25:
That was by design, so that only one is loaded into memory at the same time, so that it is speed and memory efficient.  The same thing is done internally in pyhessio: the data are always loaded into one data structure that is overwritten, so there is no copying overhead.  Basically, all the things in the Container structure are just pointers, so the data are not copied.

If you want to do what you suggest, we could create a second event source (or an option in it) that does a deep copy for the use cases when that is useful (which is not the typical use case)

```python
from copy import deepcopy
events = hessio_event_source(filename)
eventlist = [deepcopy(event) for event in events]
```

or you can create a deep-copied generator directly:

```python
copiedevents = (deepcopy(event) for event in hessio_event_source(filename))
```

then `copiedevents` should work as you expect

kosack, 2016-03-25:
I guess that should be added to the documentation, and perhaps add a second generator helper function like `hessio_copied_event_source()` or something like that (maybe suggest a better name), so that both use cases can be handled (the memory-efficient one with no copies, and a copied one for advanced use like buffering, etc)
 
Does that sound reasonable?

closed 2017-05-18 by kosack