cta-observatory/ctapipe #71: Should we process data by Chunk (set of N events) or by Event (single event)?

kosack, 2016-03-17:
Should we chunk the data or not? Here we are not talking at the file or i/o level (which is chunked by default in some optimal way), but rather for the algorithms.  Therefore it implies a major difference to the internal data structure.

#70 is an example implementation of a chunked data structure that could replace the current `RawDataContainer` structure in `ctapipe`

* *event-wise*:  
  * simple to implement algorithms
  * low memory footprint
* *chunk-wise*:
  * can make use of CPU and GPU features, since have many events in a block of memory
  * the camera trace data is a 3D array (nevnt, npix, ntimeslice)
  * simplifies some algorithms that need to loop multiple times over a set of events

However, it was pointed out that if we use a fancier stream processing system, that system can buffer events into a chunk, and provide efficient event-wise access.

mackaiver, 2016-03-17:
I agree that a chunk-wise operation can be really useful for some algorithms that need multiple events.

I think many of these algorithms need many (in the order of thousands) events to accumulate  reasonable statistics for whatever they are calculating. In that case we need huge chunks or have running statistics over many chunks. 
For the latter we could use something like  [python_runstats](https://github.com/grantjenks/python_runstats) (I haven't tried it but it looks nice.)

> However, it was pointed out that if we use a fancier stream processing system, that system can buffer events into a chunk, and provide efficient event-wise access.

I'm personally not to worried about that. As long as the chunksize is reasonably small this will work just fine. The RDD messages used in Apache Spark for example are basically "chunks".

However I think  it's important to completely hide the implementation details from the user.

kosack, 2017-05-18:
seems the answer so far is by event

closed 2017-05-18 by kosack