cta-observatory/ctapipe #76: Instrument module:  Just store camera geometries by camera id, not telescope id

MaxNoe, 2016-03-18:
This will drastically reduce time and memory usage.

Give the telescope table a column camera id and then just create tables for each camera, not for each telescope.

GernotMaier, 2016-03-18:
I don't think so: only in the MC the cameras are identical. In reality, each camera will be different.

For the MC production, already now some of the cameras are different (as e.g. the quantum efficiency is the base value + a random value calculated using the expected uncertainty)

kosack, 2016-03-25:
This bring up a point, however - the reason we break down the instrument description into multiple sub-classes/tables is for exactly this use case.

The CameraGeometry is probably fixed for all cameras (at least for a single camera version and type), but other parts of the instrument model change (like the QE, etc) and those should be in a different object/table.  That was the point of the ObsConfig data model : separate things that change from things that don't change often, and group things that change roughly at the same time, so they can be identified by a unique key. 

 Max is correct: we should be able to load a single `CameraGeometryDescription` per type, but still load a different `CameraDetectorDescription` (or whatever you want to call it) that contains the QE's, etc per camera.  In an event display you only need the fixed geometry, but in the MC you need all the info.

That implies that you need to break up the CameraDescription into at least these two pieces

MaxNoe, 2016-04-07:
Yes, I just meant `Geometry` not values that might be different for each camera.

closed 2017-05-18 by kosack