cta-observatory/ctapipe #78: White lines on GCT camera plot

watsonjj, 2016-03-23:
When plotting a GCT camera image there are unexplained white lines on the pixels (attached). This seems to also happen with other square-pixel cameras.

Is this due to the pixel positions? Camera curvature?

[image]

kosack, 2016-03-23:
Huh that wasn't there before. It could easily be due to slightly incorrect pixel spacing (too small), which could be related to the curviture. What we are plotting is the projection of the curved plane, but we do that so far by simply ignoring the Z axis. 

When you zoom into the plot around one of the white lines, do they disappear?

kosack, 2016-05-19:
It seems to be back-end dependent. Rendering to PDF works fine for example, and changing to do a different backend fixes it sometimes.   The problems seems to be antialiasing of the rectangles used to draw the camera. Maybe adjusting the spacing so there is a slightly larger gap between pixels would help?