cta-observatory/ctapipe #82: Calibration coefficients look wrong for Prod3

kosack, 2016-03-23:
Reading the peds and gains using `pyhessio`  like this:

```python
peds = pyhessio.get_pedestal(tel_id)[0]
gains = pyhessio.get_calibration(tel_id)[0]
plot(peds, gains, 'r+')
```

Works for Prod2 files:
[image]

But for Prod3 files you get 2 distinct distributions of pedestals and gains:

[image]

Looking at it in image space, here is what you get when applying the coefficients, which looks like every other pixel has the wrong calibration (implying mixing of the two gain channels):

[image]

And the raw data

[image]

kosack, 2016-03-23:
The problem appears to be that the gains for channel 0 and 1 are mixed in Prod3.  In Prod2 there is onyl one gain channel, so perhaps that is why they are not mixed.    This was noticed also by @moralejo  who then commented out the code (I reenabled it, but with a command-line switch, since it is not really a problem, just in the reading)

/cc  @jacquemier

jacquemier, 2016-03-24:
Bug corrected with pyhessio v0.5.4 (commit 61570cd5117d4822bcee182ccb8dff2eb18ddfa2)

closed 2016-03-24 by jacquemier

jacquemier, 2016-03-24:
PROD3 with pyhessio v0.5.4
[image]

moralejo, 2016-03-28:
The change related to the command-line switch in read_hessio_single_tel
does not work for me:

    parser.add_argument('', '--calibrate', action='store_true', help='apply
calibration coeffs from MC')

It seems it does not like the first (empty string) argument. It works if I
add there anything which is not already used by other switches.

A.

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

kosack, 2016-03-29:
strange - I had the same problem but thought I had checked in a fixed version. I'll fix it now.

kosack, 2016-03-29:
it seems the fix is in there already - you probably just need to sync your fork and update (or do `git pull cta-observatory master` to merge in the changes)

moralejo, 2016-03-29:
Hi,

> it seems the fix is in there already - you probably just need to sync your
> fork and update (or do git pull cta-observatory master to merge in the
> changes)
Yes, I did precisely git pull cta-observatory master... After which I got
the calibration back, but could not run it unless I made the above
mentioned change.

Should that first argument of parser.add_argument  be really empty?

A.

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

kosack, 2016-03-29:
Sorry, I just noticed I hadn't accepted the pull request (from myself!) I'm still getting used to this system. It should be there now.  The fix was just to remove the blank argument. 

In any case, we should switch to using ctapipe.config instead of argparse directly, but that's a future change

moralejo, 2016-03-29:
It works now, thanks!

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es