cta-observatory/ctapipe #98: CameraGeometry.guess

rdelosreyes, 2016-04-26:
I changed the calibration code, so that it goes faster. It took 0.2 sec to run the nb_peak_integration_mc function.

I merged the ctapipe master branch into my repository and only asked me to include the focal length as a parameter to the CameraGeometry.guess function.
After that the function now takes 1 second to run.
Timing the different parts of the code I found that calling the CameraGeometry.gues function takes 0.8 seconds, while it takes 0.2 seconds for the rest of the code.
Is it a feature or a bug?

kosack, 2016-04-26:
That sounds very slow... It could be a problem with the neighbor list building.  IN any case, you should only need to guess the camera geometry one time at the start of the "run", and then use it for every event, so maybe it's not worth worrying about?

rdelosreyes, 2016-04-27:
Solved using InstrumentDescription. It gives even better results: 0.1 sec to run now nb_peak_integration_mc function on telescopes LST, MST and SST.

closed 2016-04-27 by rdelosreyes