cta-observatory/ctapipe #131: Standardised outputs for event reconstruction

ParsonsRD, 2016-06-15:
In pull request #129  I added a first pass at a reconstruction module. However, at some point we should try to come up with a standard-ish set outputs for such classes. 
I would guess this means coming creating some kind of "reconstructed shower" container that looks similar to that for the Hillas parameterisations.

At a first guess I would say this should hold the reconstructed sky coordinates of the shower (in Alt, Az?), the core position (in the ground system), estimated energy, and xmax. Probably the uncertainty on all these values should be included as this is available in some analyses.

kosack, 2016-06-15:
Yes, that sounds like a good idea. And do include uncertainty - I think that's even in the requirements or specs somewhere (or at least was heavily discussed).  We want all reconstruction methods to at least try to estimate an uncertainty.

This brings up a main point about data structures: what should we be using for things like parametersizations (Hillas, reconstructions, gamma-hadron parameters, etc)?

* a bare class (perhaps with __slots__ set, so that users cannot add new data)
* a `ctapipe.core.Container` subclass
* a simple dict (but then we have to require the item names to be somehow consistent)
* a `namedtuple` (compact an immutable)
* a C struct (which you can create and use via python using the standard library `struct` module)

Some use cases/requirements:
* should be easy to pass the data to a C/C++ routine (or cython) if we want to write more optimized code later

ParsonsRD, 2016-06-15:
I created a simple example here #132 using a named tuple, but this may not have enough functionality for what we ultimately want to do with this. For pros and cons of each I don't have a good feeling for what would be best to go with. Can we assign a few people to the conversation here who may have more of a feeling for this?

ParsonsRD, 2016-10-12:
This is a first pass at the variable we may need in the shower objects. Comments welcome!

**Common output format for reconstruction**

- Container class for storage of shower parameters, need to standardise output for shower reconstruction methods.
- Alt, Az - storage
- Uncertainties - absolute positional error or per parameter? (maybe we could store all 3)

- Core position - in x and y
- Uncertainties absolute error is probably OK

- List of used telescope IDs
- Sum of charges on selected telescopes (average size)
- Goodness of fit, quality selection parameter (could be a list or dict)

- Xmax and uncertainty

- bool is_valid? fit_status etc for book keeping purposes

- Some kind of meta data, where this object came from, what reco code etc

**Separate energy output**

- Energy and uncertainty
- Tel IDs used in energy reconstruction somehow
- Goodness of fit

- meta data about algorithm and should probably know where shower object came from
- bool is_valid?

TarekHC, 2016-10-13:
Following this discussion, we created the `shower_reco` branch, and a pull request (#202) with a first direction reconstruction container and algorithm base class.

closed 2016-11-21 by ParsonsRD