cta-observatory/ctapipe #137: setup.py not valid for  pytohn 3.5.2

jacquemier, 2016-07-06:
After update my python to 3.5.2 (unintentional), the "make doc" command does not work.
```
(cta)[jacquem@lappc-p415] $ make doc
CTAPIPE_EXTRA_DIR=/home/jacquem/workspace/CTA/ctasoft/ctapipe/ctapipe-extra python setup.py build_sphinx
Traceback (most recent call last):
  File "setup.py", line 25, in <module>
    conf = config.ConfigParser()
AttributeError: module 'distutils.config' has no attribute 'ConfigParser'
Makefile:38: recipe for target 'doc' failed

```
It seems like distutils package is not backward-compatible with python 3.5.1

kosack, 2016-07-06:
That's surprising! But I guess we should fix both a minimum/maximum version of python for each release anyhow, rather than trying to support bleeding-edge versions. I'd guess distutils will catch up soon.

MaxNoe, 2016-07-06:
distutils switched from using `configparser.ConfigParser` to `configparser.RawConfigParser`.

You should just use either of them directly, not through `distutils`.

commit 6988b64c (2016-07-06): Merge pull request #138 from MaxNoe/master

Fix setup.py, fix #137
commit e01f5f80 (2016-07-06): Fix setup.py, fix #137
closed 2016-07-06 by kosack