cta-observatory/ctapipe #148: InstrumentDescription interferes with event container and pyhessio commands

watsonjj, 2016-07-14:
When loading an InstrumentDescription object, this reopens the simtel file, gets the information in needs and then **closes the file**. Now all pyhessio functions have no simtel file open to extract data on, and they fail. Additionally calling event = next(source) will also fail.

Even if you removed the close_file() from ID.load_hessio() - because the simtel file has been reloaded, the eventstream is reset to the first event in the file, not necessarily the event you are expecting the pyhessio functions to extract from. Then calling event = next(source) will give the second event in the file, again, not necessarily the event you are expecting. 

The only way to make this work is to load the InstrumentDescription before ever calling hessio_event_source() first, and making sure you don't load the InstrumentDescription again afterwards.

Below is a small script to display this issue.

watsonjj, 2016-07-14:
`from ctapipe.utils.datasets import get_path
from ctapipe.instrument import InstrumentDescription as ID
from pyhessio import get_teldata_list

filename = get_path('gamma_test.simtel.gz')
source = hessio_event_source(filename)
event = next(source)
event = next(source)
event = next(source)
print(event.dl0.event_id)
print(get_teldata_list())
tel, cam, opt = ID.load(filename)
print(event.dl0.event_id)
try:
    print(get_teldata_list())
except:
    print("ERROR RAISED")
    pass
try:
    event = next(source)
except:
    print("ERROR RAISED")
    pass
try:
    print(event.dl0.event_id)
    print(get_teldata_list())
except:
    print("ERROR RAISED")
    pass`

watsonjj, 2016-07-14:
```
from ctapipe.utils.datasets import get_path
from ctapipe.instrument import InstrumentDescription as ID
from pyhessio import get_teldata_list

filename = get_path('gamma_test.simtel.gz')
source = hessio_event_source(filename)
event = next(source)
event = next(source)
event = next(source)
print(event.dl0.event_id)
print(get_teldata_list())
tel, cam, opt = ID.load(filename)
print(event.dl0.event_id)
try:
    print(get_teldata_list())
except:
    print("ERROR RAISED")
    pass
try:
    event = next(source)
except:
    print("ERROR RAISED")
    pass
try:
    print(event.dl0.event_id)
    print(get_teldata_list())
except:
    print("ERROR RAISED")
    pass
```

watsonjj, 2016-07-14:
The output you get from this is:

> I/O block extended by 385996 to 1385996 bytes
> WARNING: ErfaWarning: ERFA function "taiutc" yielded 1 of "dubious year (Note 4)" [astropy._erfa.core]
> gzip: error writing to output: Broken pipe
> I/O block extended by 744748 to 1744748 bytes
> 803
> [ 17 104 124]
> Hessio file will be opened.
> gzip: error writing to output: Broken pipe
> I/O block extended by 744748 to 1744748 bytes
> Astropy tables have been created.
> gzip: error writing to output: Broken pipe
> Hessio file has been closed.
> 803
> ERROR RAISED
> ERROR RAISED
> 803
> ERROR RAISED

watsonjj, 2016-07-14:
If you remove the close_file() from ID.load_hessio() instead you get:

> I/O block extended by 385996 to 1385996 bytes
> WARNING: ErfaWarning: ERFA function "taiutc" yielded 1 of "dubious year (Note 4)" [astropy._erfa.core]
> gzip: error writing to output: Broken pipe
> I/O block extended by 744748 to 1744748 bytes
> 803
> [ 17 104 124]
> Hessio file will be opened.
> gzip: error writing to output: Broken pipe
> I/O block extended by 744748 to 1744748 bytes
> Astropy tables have been created.
> 803
> [38 47]
> 409
> [ 11  21  24  26  61  63 118 119]
> gzip: error writing to output: Broken pipe

watsonjj, 2016-07-14:
What I would have expected the numbers in output should be is:

> 803
> [ 17 104 124]
> 803
> [ 17 104 124]
> 4907
> [ 1  2  4 14 15 17 19]

i.e. loading the InstrumentDescription shouldn't interfere with the eventstream for the event container or the pyhessio functions. I understand this may be hard to achieve, but I guess this is a similar reason global variables are frowned upon.

kosack, 2016-07-18:
This is a fundamental problem with pyhessio, which keeps a global state for the currently open file (so only one file can be opened at once, and calling open 2 times causes a conflict as you show).  That was basically just due to how the `read_hess.c` code from simtelarray worked, but it maybe could be solved by making pyhessio object-oriented, or to have it keep a stack of `AllHessData` structs (one per open file handle).  It is therefore a problem in the underlying C code...

This also points to the fact that the current use of the instrument module needs a lot of work to be a bit friendlier

kosack, 2016-07-18:
I guess a current workaround is to load the ID info first, and only afterward, use the HESSIO source functions.

closed 2017-05-18 by kosack