cta-observatory/ctapipe #150: Definition of angles in hillas.py

ParsonsRD, 2016-07-20:
Currently the definition of the 2 angles reconstructed when determining the Hillas parameters is a little confusing. There are 2 angles created:

The rotation angle of the image centroid wrt the camera centre (in my mind this should be called **psi**)

The rotation angle to the actual image (I would usually call this **phi**)

Currently they are defined the other way round (I'm not saying this is wrong, just want to make sure we are consistent), but also the second parameter (phi in my naming scheme) is not created for hillas_parameters_1 and this should really be added if this function is to be useful.

kosack, 2016-07-21:
feel free to rename them - there is no standard right now, so it's a good time to pick better names!  Already phi and psi are confusing, since we also use them for offset angles, but at least making them more consistent will help.  The difference comes from what was considered standard in Whipple vs HESS, which were not the same.

mdpunch, 2017-01-05:
Strange, thought I put a comment here, but it disappeared!

@ParsonsRD @karl, I don't think there's a standard in Whipple vs HESS (and strange that there's a difference, probably Whipple vs. HEGRA).

My (Whipple) thesis agrees with Dan (or vice versa), *phi* being the angle between the x-axis and the centroid (so `atan2(y,x)`) and *psi* the angle between the x-axis and the image major axis, but taking into account the quadrant that the centroid is in (so giving a comprehensible sign for skewness/asymmetry, +ve with the tail pointing away from centre).

I've noticed that some routines take the angles defined with respect to "up".  Is there any reason to do this?  Trigonometrically, it's weird, and leads to pi/2's being added to Hillas routines.  Can we stick to angles being w.r.t. positive x-axis?

kosack, 2017-01-05:
I have no idea why anyone would use up as the angle definition... the x-axis makes more sense to me, so let's just standardize on that.

kosack, 2017-01-05:
Checking my whipple code, I had the same x-axis definition, and I think HESS does too.

mdpunch, 2017-01-05:
Sounds good to me!  Is there somewhere to put these standards (an interface document??) so that people can have a reference (and eventually give reasoned disagreement, if they want)?

closed 2018-12-19 by MaxNoe