cta-observatory/ctapipe #173: DL2-like data: columns and format

TarekHC, 2016-10-03:
Hi all, 

I am creating a python exporter for the Prod-3 Mars analysis files. The idea is to export the result of our analysis to a "DL2-like" format, to develop the latest steps of the MC pipeline (IRF calculation and DL3 construction, basically).

After speaking a bit with @kosack , it seems that a normal astropy.table may be enough for the DL2 data level (a table containing the relevant reconstructed event quantities). But as far as I understand, DL2 data may (and probably will) have reconstructed parameters from different analysis chains.

Should I just create a simple table (in FITS, for example) with the parameters/metadata I can think of, and improve it with time? Or should we try to give it some more complexity (new class/container to which each analysis may feed their resulting reconstructed quantities (e. g. as defined in #131 ), etc...)?

kosack, 2016-10-03:
I didn't quite understand that you were making an exporter for MARS and not a writer for `ctapipe` . In that case I agree you should start with just making an example FITS table as you suggest, and we can discuss how to implement the export in `ctapipe` and MARS. The contents are more important. 

Also, You should start with the DL3 format specification.  An `astropy.table.Table` should work fine for your purposes since Dl2 is just DL3 with extra columns.  

The main issues are the following:

* DL2 files should look very similar to DL3 files  (a FITS table with 1 row per event, with an `EVENT_ID` and `TIME` column followed by columns for each parameter)
* DL2 files contain the shower reconstructions and "hadronness" parameters from the pipeline for all events *before* any gamma-hadron separation, event-classificaition, or method-selection has been applied. There is no per-telescope information, other than the trigger pattern.
* DL2 files may contain multiple shower reconstructions and multiple measures of gamma/hadronness
  *  ideally they should all be in one table (rather than one table per algorithm). The reason is that they will be fed into various machine learning algorithms simultaneously to classify events and select the best final reconstruction parameters. If they are in separate tables, a way to make a *join* operation would be needed that can change the column names to keep repeated columns unique, which could be more complicated. 
  * A single table implies a column-naming standard to keep track of which measured quantity came from which algorithm. 
  * The column naming standard could be as simple as prepending (or appending) the algorithm name to the DL3 column definition (`ALG1_ENERGY', `ALG1_RA`, `ALG1_DEC`, `ALG2_ENERGY`, `MYALG_ENERGY`...) Where `ALGx` may be an identifier for that algorithm.)
* DL3 is essentially a subset of DL2 that only contains the final shower reconstruction and a single hadronnness parameter, with no pre-pended algorithm names. It additionally contains the results of the classification or event type. It may for instance contain the ENERGY from ALG1 and the RA/DEC position from ALG2, etc.

*a side note*:  Doesn't MARS already have a FITS event-list exporter? I know in the original "mini data challenge" we had data from Magic that came from it, so you might check first. It may just need modification to include more columns.

TarekHC, 2016-10-03:
Hi @kosack , thanks for the quick reply,

> I didn't quite understand that you were making an exporter for MARS and not a writer for ctapipe

Well, its closely related. First I wanted to provide reconstructed data that people (and me) could use for development. It will be just temporal, until we have a working chain within ctapipe (which may take a while). If you think it won't be that useful, or think there may be an easier way, please let me know! :smile: 

In my mind, as soon as we have that, we can develop IRF calculation, method evaluation, DL2/DL3 writers...

>The column naming standard could be as simple as prepending (or appending) the algorithm name to the DL3 column definition 

I was thinking exactly the same thing. That's why it would make sense to use whatever standard people is using to store the output of each algorithm (a container with all reconstructed variables + metadata defining each analysis?) as input for a simplified class that creates the DL2. That way the exported values would be defined as "MARS_ENERGY", etc... Although for now just a FITS table may be easier.

Question: What about MC columns? Do I simply add them with "MC_ENERGY", etc? They will only be added once, as they are common to all algorithms, so that would be the easiest convention. The only problem is that "MC_" would look like another algorithm... Maybe "TRUE_*"? Also MC files will be required to contain the metadata describing how they were generated (# simulated showers, slope, etc...), which is not defined within the DL3 specs.

> a side note: Doesn't MARS already have a FITS event-list exporter? I know in the original "mini data challenge" we had data from Magic that came from it, so you might check first. It may just need modification to include more columns. 

I am actually working on that from the MAGIC side... Updating that exporter, which is pretty much deprecated. Not much code can be re-used (it is pure C++), but at least we have the know-how.

kosack, 2016-10-04:
> Question: What about MC columns? Do I simply add them with "MC_ENERGY", etc? They will only be added once, as they are common to all algorithms, so that would be the easiest convention. The only [...]

Yes, I think you're right that just adding them as columns is the best way.  I'm not sure about the naming convention... Either TRUE or MC  or SIM is fine with me.  By the way, those columns may also end up in DL3, so as long as we are consistent it's ok.  It's true that the monte-carlo columns will be a bit different since they may have more physical parameters (e.g. there is a `MC_XMAX`, but not all reconstructions will calculate an X_MAX), and the particle type will have to be there as well (`MC_PARTICLE_ID`) and that also needs a convention  for numbering which is a gamma, proton, electron, higher-Z.  That could be something like  using the number as the Z of the particle: 
* -1=electron
* 0=gamma 
* 1=proton 
* 2 = helium nucleus
* 3 = Z=3, ...

maxnoe, 2020-10-26:
I think we can close this for now. The DL2 data format will be (for now)  similar to the dl1 format, with the fields changed respectively.

closed 2020-10-26 by maxnoe