cta-observatory/ctapipe #174: Container class write method not working for hierarchical container structures

TarekHC, 2016-10-04:
Hi all, 

Within ctapipe, most data containers are created by constructing hierarchical structures. Example:

```python
container = Container("test")
container.add_item("dl0", RawData())
container.add_item("mc", MCEvent())
```

The container class "write" method works properly as long as no hierarchical structure is given:

```python
from ctapipe.core import Container
from astropy import units as u

container = Container("test")
container.add_item('energy')
container.energy = 3.5 * u.TeV
print(container.to_table())
container.write("test.fits",format="fits")
```

But it does not work in the following case:

```python
from ctapipe.io.containers import RawData, MCEvent, MCShowerData
from ctapipe.core import Container

container = Container("test")
container.add_item("mc", MCShowerData())
container.mc.energy = 3.5 * u.TeV
container.mc.alt = 1 * u.deg
container.mc.az = 1 * u.deg
container.mc.core_x = 0 * u.meter
container.mc.core_y = 0 * u.meter
container.mc.h_first_int = 10000 * u.meter
print(container.to_table())
container.write("test.fits",format="fits")
```

Well, to be more specific, the "to_table()" fails while checking the type of each element (MCShowerData() has no dtype).

As most containers I have seen use this structure, we should probably improve the "to_table()" function.

kosack, 2016-10-04:
The `to_table()` conversion is still pretty experimental and not ready for use.  We did mention last time that one way to do it properly may be to "flatten" the structure first, so that if you have something like a sub-table, you would just append extra columns with a tag prepended to their name. Alternately you just write all sub-tables to separate tables with a foreign key index (like in SQL).

The whole output scheme needs thought though - this may not be at all efficient the way it is implemented, especially since we want to write one row of a table, not the full table at once.

kosack, 2016-10-04:
Really we may just need a Component that takes a Container as input. The first time it gets it creates a FITS table as output, and each subsequent time writes a row to that table.  That may be more general and easy to use than embedding the conversion functions inside Container.  

So you would then write data like this:

```python

with FITSTableWriter("output.fits", "EVENTS") as writer:
    for container in data:
        writer.process(container)

```

elehcim, 2016-10-13:
First implementation here #212