cta-observatory/ctapipe #177: Logging

MaxNoe, 2016-10-06:
I often see stuff like

`log.info('[File] something happend with file')`

I think this not best practice with the logging module. The logging module provides several ways to provide information or categories. Putting some markup style in the message is certainly not the best one.

kosack, 2016-10-06:
Yes, The correct way to log now is to use the logger registered in a `core.Component`:
  `self.log.debug("debug message")`

That logger automatially becomes a sub-logger of the parent when used in a Tool, and
The file name and other metadata are automatically appended in the default format.

Module writers need to migrate to use this method. I will bring that up at the meeting.

MaxNoe, 2016-10-06:
There should also be proper logging in modules not using a `core.Component`.

This should be done via instantiation the proper logger at the top of every file like this:

````
import logging

log = logging.getLogger(__name__)
```

cdeil, 2017-04-22:
At  https://cta-observatory.github.io/ctapipe/development/code-guidelines.html#logging-and-debugging it says
> do not use print() statements to output text. Instead use the common logging failities of ctapipe

but there's no info what those common logging facilities are. @kosack - Can you add some info how logging in ctapipe works?

Also -- I would suggest to separate out debugging into a separate section in the docs. The debugging section can reference the logging section as one way to debug, but it's quite separate.

closed 2017-06-07 by kosack