cta-observatory/ctapipe #179: Remove all configuration parsing from `calib` module

kosack, 2016-10-06:
The calib module current handles its own configuration parameters, which is bad. It should use the functionality in `ctapipe.core.Component` (for algorithms) and `ctapipe.core.Tool` (for executables). No parameters should be passed as bare `dicts` as is currently done. 

A cleaner class hierarchy should be created to separate configuration of various sub-algorithms of calibration like the choice of trace integration method. Those classes should use `Component` functionality to define user-choosable parameters.

watsonjj, 2016-10-06:
I am happy to develop the `calib.camera` module to use this functionality.

kosack, 2016-10-06:
Great! That would  a good test of the common functionality for configuration, which is so far only used in a few tools. If you have problems or suggestions to make it nicer, those are welcome. 

For the camera code, I would suggest identifying the various sub-algorithms and making a sort of base-class the derives from `Component` for each that defines a common interface. For example, a trace integrator can be thought of as a black box that takes a pixel data cube (2D array of pixels vs time) and returns a set of 1D arrays with the integral and maybe other timing parameters.  Don't be too fancy with the class structure, since we want things to be as simple as possible, but the `Component` functionality can help to group functions with their associated user parameters.

something like the following (though you can decide on the exact interface):

```python
class Integrator(Component):
     """ common interface for all trace-integration routines """
    def integrate(traces):
        """ all Integrators should define the integrate function that returns a common value """
        return traces.sum()

class FixedWindowIntegrator(Integrator):
    start_sample = Integer(...)
    end_sample = Integer(...)

    def integrate(traces):
        return traces[self.start_sample, self.end_sample].sum()

```

Then you can create a class like `Calibrator` that delegates integration to an `Integrator` instance:

```python
calib = Calibrator(integrator=FixedWindowIntegrator())
calib.calibrate(event)
```

If things work correctly, and Calibrator contains an Integrator, the user options like `start_sample` can be configured from a config file or command-line.

watsonjj, 2017-02-28:
The calibration has been updated to be Components, and no longer pass configuration around in a dict.

closed 2017-02-28 by watsonjj