cta-observatory/ctapipe #188: Configuration module uses status codes

MaxNoe, 2016-10-11:
The configuration modules return status codes. 

They should raise exceptions instead. Status codes are kind of eighties.

jacquemier, 2016-10-11:
Which configuration module ?
ctapipe.util.json2fits ?

MaxNoe, 2016-10-11:
Yes.

MaxNoe, 2016-10-11:
They do stuff like

```{python}
try:
    stuff()
except IOError:
    return False
return True
```

I do not think this is a good idea. Also for logging exceptions, there is `logging.exception` which includes the traceback.

```{python}
try:
    stuff()
except:
    logging.exception('Could not do stuff')
    raise
```
Would log the exception including traceback and the raise the same exception. So only logging is added.

jacquemier, 2016-10-13:
https://github.com/cta-observatory/ctapipe/pull/196

closed 2016-10-25 by jacquemier