cta-observatory/ctapipe #195: add API for writing FITS header cards with units attached automatically

kosack, 2016-10-13:
```python
io.write_fits_header(fits, key, value, help)

io.write_fits_header(fits, key='E_MIN', 10*u.TeV)
io.write_fits_header_from_dict( fits, headers )
```

Does this already work with `astropy.table.QTable`? or do we need to add a function?

facero, 2016-10-13:
To keep track, right now we do:

```python
def quantity_to_tuple(qt,unit_str):
    """
    Splits a quantity into a tuple of (value,unit) where unit is FITS complient.
    Useful to write FITS header keywords with units in a comment.
    Parameters
    ----------
    qt : astropy quantity

    unit_str: str
        unit string representation readable by astropy.units (e.g. 'm', 'TeV', etc)

    """
    return qt.to(unit_str).value, qt.to(unit_str).unit.to_string(format='FITS')

metadata['mc_energ']= quantity_to_tuple(event.mc.energy,'TeV')
save_fits(img, output_file_path,metadata)
````

Where in save_fits we do something like:

```python
val=(value,quantity)
hdu_list[0].header[key]=val[0]
hdu_list[0].header.comments[key]=val[1]
```

maxnoe, 2020-10-26:
This is solved at least for table io using the serialize meta option in astropy 4

closed 2020-10-26 by maxnoe