cta-observatory/ctapipe #204: json vs yaml

jacquemier, 2016-10-13:
Many ctapipe developers want to use yaml configuration instead of json.
I proposed to list the pros and cons here,  in order to make an informed decision

JSON:
    PROS:  
        -  json library is included inn  the Python Standard Library (https://docs.python.org/3/library/json.html)
        - ctapipe uses traitlets as configuration system, which uses json for configuration file (no support for yaml)
    CONS:
        - Comment are not allowed in JSON
        - braket syntax is no so readable.

MaxNoe, 2016-10-13:
There is a great talk on serialization formats from PyCon 2015:
https://www.youtube.com/watch?v=kjZHjvrAS74

TL;DR:
Json or yaml doesn't really matter, json has more boilerplate but is safer. Yaml can be dangerous if read in with `yaml.load` instead of `yaml.safe_load`.

Note:
The `pyyaml` module is currently unmaintained and has a long standing bug concerning the parsing of scientific notation floats:
http://pyyaml.org/ticket/246

Conda switched to this fork:
https://pypi.python.org/pypi/ruamel.yaml

Which is actively maintained and resolved the bug mentioned before.

MaxNoe, 2016-10-13:
FYI:
```
In [1]: import yaml

In [2]: yaml.load('1e3')
Out[2]: '1e3'

In [3]: import ruamel_yaml

In [4]: ruamel_yaml.safe_load('1e3')
Out[4]: 1000.0
```

mackaiver, 2016-10-14:
If we decide on the `yaml` format, we should go for the `ruaml` lib. 
While the json format does not allow comments there is a workaround(its quite ugly)

```

In [1]: import jsmin

In [2]: jsmin.jsmin('{\n "bas":223 //some comment we can put here \n}')
Out[2]: '{"bas":223}'

```

MaxNoe, 2016-10-14:
Oha, this is really hacky.

dneise, 2018-12-11:
The core developers on the framework sprint decided to got for YAML.

closed 2018-12-11 by dneise