cta-observatory/ctapipe #225: clean up Hillas parameters code

kosack, 2016-10-16:
Currently there are two functions: `ctapipe.image.hillas_parameters_1` and `ctapipe.image.hillas_parameters_2`

The following problems should be fixed: 

* The version 2 code is simpler and easier to read and the implementation is faster, but is not used by default (though it was in the past). Suggest either modify this version to be the "official" one, or rewrite version 1 in a cleaner manner
* higher-order moments are implemented so far in version 1 only
* the rotation angle parameter does not agree between the two (matter of definition of angles?) check
* skewness and kurtosis are calculate by hand in version 1, instead of using `scipy.stats.skew()` and  `scipy.stats.kurtosis()`
* both do a lot of linear algebra by hand, and could benefit from using standard linear algebra routines in `scipy.linalg` rather than calculating cosines and sines manually (though this is not a major issue)
* the speed of version 1 could be improved  by using numpy more extensively as in verison 2, or using *cython*

wrijupan, 2016-10-18:
- I tried to implement higher order moments in version 2 : #193.
- If directly used scipy.stats.skew() and scipy.stats.kurtosis(), it does not allow to put weights to the data array. Is there a scipy function which allows us to do weighted statistics? I could not find any, so for now, I put weights by hand.
- The rotation angle disagreement is because of units. This problem is now resolved in #193

wrijupan, 2016-10-18:
- Also currently, none of the two versions returns azwidth although they calculate it. There is inconsistency in azwidth value between the two versions.

kosack, 2016-10-18:
I'm not sure anyone uses *azwidth* for anything, so maybe we can just remove it? Unless there is some reconstruction algorithm that needs it, but I'm not aware of one.

wrijupan, 2016-10-19:
I have removed the azwidth calculation from the hillas parameters code, unless there is a strong objection to keep it. Also I have removed S_x4 and S_y4 calculations from version 1, since it was not being used anywhere in calculations.

closed 2018-06-19 by kosack