cta-observatory/ctapipe #226: pyhessio broken pipe and seg faults

moralejo, 2016-10-16:
The hessio_event_source generator gives the message "gzip: error writing to output: Broken pipe" whenever the generator is **not** used up to the last event in the file.

This can be reproduced, e.g. by executing the following in ctapipe/io/tests  :
py.test -s test_hessio.py

Can this be fixed? 

Besides, at least in my system (OS X 10.11.6), pyhessio gives "Segmentation fault: 11" quite frequently (at least in ~10% of cases executing test_hessio.py). Is this also the case in other systems?

jacquemier, 2016-10-16:
This test uses gamma test.simtel.gz which has been truncated by Karl to have a reasonable size. Last data are not correct.

"Segmentation Fault: 11" does not happen on my Linux.
Could you try to get more feedbacks thank to a debugger ?

moralejo, 2016-10-16:
I am aware of the truncation of the file, and it explains the message "read xxx bytes and expected yyy" that is also displayed often. It would certainly be better to have a proper short file. However, I doubt the "broken pipe" message is related to that, because it is precisely when one does **not** read the full file that the message appears.

Good to know you do not get the seg fault in linux. How can I debug that, given the problem is most likely in the c-code ?

jacquemier, 2016-10-17:
hessio code ( from Konrad) uses gzip internally to unzip file piece by piece. 
As far as I know, It is not foreseen to not  reading the full file.
This is why "gzip: error writing to output:" message appears.
Unless this is a blocking issue, I can not waste time to modify hessio C code, which  will disappear at soon as DL0 format will be chosen.

jacquemier, 2016-10-17:
Concerning "Segmentation Fault: 11",  is it a blocking issue or does it only appears at end of execution ?

moralejo, 2016-10-24:
Hi, a previous reply made via e-mail has got lost (i.e. has not been added to this thread) - any idea why?

The seg fault, when it happens, indeed interrupts the execution, for example, of the test module test_integrators.py. 

I tried to use as input two other simtelarray files (instead of the default test file gamma_test.simtel.gz), and I could not reproduce the seg fault with any of them. On the other hand, the default test file which makes trouble with pyhessio never makes the seg fault when I run over it (outside ctapipe) read_hess or the Chimp converter. Very strange.

Cutting the last 635256 bytes of the test file (which is a "brutally truncated" one) makes the message "read xxx bytes and expected yyy"  disappear, as expected, but it has no effect on the seg fault (nor on gzip's broke pipe message).

Being a rare issue, happening just on Macs and on one file, I think we can forget about it for now.

moralejo, 2016-10-25:
The gzip thing is not a blocking issue, so we can live with it. But the seg
violations are;  in my system they make the tests which use pyhessio fail
quite frequently.  I have never had such error using hessio in other
programs, like e.g. the chimp hessio-to-MARS converter. If there is some
memory mismanagement it may be that even when no seg violation occurs the
program is not reading all the data properly.
I will try to understand the problem better.

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

kosack, 2016-10-26:
I think it's not the truncated file: The segfaults are not really part of pyhessio, but due to the underlying library libhessio, which only allows one file open at a time.  Currently in some of the tests, 2 files are opened without closing the first (once for reading the telescope layout and once for the event loop), which causes a segfault when the two conflict. 

Either we need to prevent that from happening, or somehow make libhessio reentrant (which seems to be a very large amount of work).  This is only a temporary problem, however, and only occurs if the instrument module is used to read the focal length manually (rather than using the data container and `hessio_event_source`).  

Once we have an instrument database, we won't need to read it from the simtelarray headers, and this should not be an issue.  For now I will open an issue to try to fix the tests and other code to not use the instrument module's hessio reading feature.

moralejo, 2016-10-26:
Funny, my lost comment from 4 days ago appears now as posted 19 h ago (?)

Thanks for looking into this. Is the 2-open-files explanation  consistent with the fact that with two other simtelarray files I have never had the seg fault, after running many times?

kosack, 2016-11-04:
I found the problem: it is not to do with 2 files open (that is also a potential problem, however). The problem is lack of ram! You need something like 8gb of ram to use `libhessio`, which for some reason does not allocate memory dynamically. Therefore if the ram is not free (and virual mem cannot be used), it simply segfaults.  

For me, closing my browser stops the crash. 

Is there a way to test for this failure to allocate global variable when loading a C-library dynamically?

moralejo, 2016-11-04:
Some info: for me the error is quite random with the ctapipe gamma test
file - but does not happen at all with the other simtelarray files I tried.
Furthermore, for the  gamma test file it does not occur either when I use
the hessio library in other programs. Not sure lack of memory is consistent
with this.

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

kosack, 2016-11-07:
It could be because it is a very early Prod2 file, with a larger super-array and larger memory usage, however it may have something to do with it being truncated mid-event. I think in any case we should abandon this broken test file and make a good one. Is there anyone who could run the MC and produce a file with e.g. only the CTA-South full array and ~10-20 events? Ideally something that is not too large, so it can be distributed with the code (10-20 MB would be ideal).

That should solve a lot of these problems, which are definitely strange and annoying.

jacquemier, 2016-11-10:
To avoid using pyhessio (wich is memory-consuming)  during unit tests, I propose to use a gzipped pickled file built thanks to the new serializer module.

moralejo, 2016-11-10:
I may be misunderstanding (sorry if that is the case), but wouldn't that be
like giving up on the hessio format as input for ctapipe?  We should aim at
being eventually able to process the full existing MC libraries with
ctapipe, I think it would be extremely useful.
Since the error I get (occasionally) does not seem to occur to anyone else,
I do not think we should put more time into it, unless it pops up again in
the future.

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

kosack, 2016-11-10:
we can also just mark tests that are slow as follows:

```python
import pytest
@pytest.mark.slowtest
def test_function():
   pass
```
Then the user can run them optionally.

By the way, I do have the same problem as Abelardo occasionally (on both my Linux and mac machines), so we should at the bare minimum provide a better example file. So far, I'd prefer to keep it as a simtel file, since the Container classes may evolve in time as we restructure things, which would break reading the pickle file.

jacquemier, 2016-11-10:
We will not giving up on the hessio format as input for ctapipe.
I just propose to use another file format for unit test.

kosack, 2016-11-10:
I've asked Gernot to produce a better sample file, in any case. Even if we use a different format for some unit tests, we need a unit test for the hessio_event_source code that reads a simtel file, so we can't avoid using pyhessio in unit tests completely.

jacquemier, 2016-11-10:
I agree

kosack, 2016-11-10:
What we maybe should do is to improve the mock_event_source code, which is not so useful as it is, and have it be able to generate a reproducable fake event that has the same structure as a real one, and that can be used for unit tests for calibration, etc. Most don't need real data for testing, but are better with something fake that is well-understood.

kosack, 2017-06-19:
I'm closing this now. The solution was to increase the stack size limit (to avoid crash in libhessio) using `ulimit -c <SIZE-IN-MB>` to at least 16384, or largest you can get on your machine

closed 2017-06-19 by kosack