cta-observatory/ctapipe #243: Coordinates for muon impact parameter

MaxNoe, 2016-10-24:
The muon fitting method implemented by @AMWMitchell returns x and y coordinates of the impact.

The method implemented by @pcumani and me returns the radius and the polar angle of the maximum.

What is preferable? What are the transformations needed to get from one to the other?

I chose the polar representation, because the radius is the value that enters most calculations concerning light intensity.

rdelosreyes, 2016-10-24:
If the direct output from the fit is X,Y, I would like to save them for debugging/verification purposes. 
But if, as you say, the radius is heavily used we can think about storage it, so that it does not have to be calculated every time, although I do not see that this calculation is so heavy computing demanding, specially considering coordinate transformation libraries, isn't it?

AMWMitchell, 2016-10-24:
The ring fitting algorithm returns the X and Y coordinates of the centre of the ring, at the moment in camera coordinates, as well as the radius. 

The intensity fitting algorithm returns the impact parameter only (not X and Y), and does not return an angle to the maximum. 

Are you sure that our code returned the X and Y of the impact position? For the impact, the impact parameter (radius of impact) is usually more important than X and Y, as long as the mirror is near symmetric. For the circle fitting, the X and Y of the centre of the ring in the camera as well as the ring radius are useful.

The transformation between coordinates (camera frame to nominal, for example) can be done using  the coordinates class: ctapipe/coordinates
A polar/cartesian transformation can probably be done with numpy.

MaxNoe, 2016-10-24:
I was talking about the members of the `MuonIntensityContainer` class. I wanted to update my code to return this Container. It has the members

https://github.com/cta-observatory/ctapipe/blob/muonmaster/ctapipe/io/containers.py#L219

```
    impact_parameter_pos_x, impact_parameter_pos_y:
        position on the mirror of the muon impact
```

AMWMitchell, 2016-10-24:
These are optional; with these classes, you don't have to fill in all parameters, as long as you fill in the *main* results of the fit. We should make more clear which parameters are those absolutely necessary.

kosack, 2016-10-24:
Just remember that what you put in the container is what ends up in the output files as columns. So it could be useful to have both representations (similar to having ALT/AZ and RA/Dec coords in the event list file, even though you can compute one from the other).  Of course transforming from cartesian to polar coords is pretty simple, so maybe just pick a standard.

MaxNoe, 2016-10-24:
Ok. But then I would just give each "operation" it's own container. There should be imho no columns which are *always* nan or empty.

This is how IceCube does it, for example. And it works quite well for them from what I hear from our icecube folks.

kosack, 2016-11-04:
I think Max's idea is good. Optional columns should have their own data structure if there are many of them. The intenion of the data writer code will be to be able to do something like this:

```python
writer = Serializer(list_of_containers)
for event in source:
    writer.write(list_of_containers)
```

where the list of containers could be a single container (basic muon parameters) or several, in the case that an algorithm creates more information.  All containers in the list will end up as columns in the output (either merged into one table with the name prepended, or as separate tables, depending on what the user wants). Therefore, having muon code produce several outputs is fine.

closed 2020-10-26 by maxnoe