cta-observatory/ctapipe #244: Remove code using instrument module to read camera def

kosack, 2016-10-26:
The issue with segfaults in #226  is likely due to the opening of 2 simultaneous simtel files (which the underlying libhessio.so library does not support). 

Any code that needs to read the camera defs should not use the  `instrument` module until it is redesigned. 

`io.CameraGeometry` should be sufficient, and it should only be constructed from the data in the `EventContainer` generated by `hessio_event_source`, not via `pyhessio` functions directly. 

This is causing very unstable code, that fails randomly on macOS machines in particular, and occasionally on linux (due to different memory handling likely)

kosack, 2016-10-26:
This particularly affects the test code in the calib module which must be re-written

tino-michael, 2016-11-10:
are we sure that the dictionary in `_npix_to_type` is correct/complete?
particularly for ASTRI cameras I found a rotation of −90° and not the given 0°
the attached plots show the "error" on the tilt angle of the hillas parametrisation for my rotation (top) and the no rotation (bottom)

[image]
[image]

kosack, 2016-11-14:
No, I think the `npix_to_type`  array was simply reverse-engineered duing the Prod-2 production, and almost certainly needs updating.

closed 2017-05-18 by kosack