cta-observatory/ctapipe #248: Single pixels in Cleaned Images

ParsonsRD, 2016-11-03:
As mentioned at the pipeline meeting, currently the split level image cleaning algorithm is implemented in such a way that it is possible to have single pixels with no significant neighbours. Although it is not a bug as such is can cause problems when setting aggressive cleaning thresholds.

It would also be nice to allow the HESS-style option where it is required that each pixel have a significant neighbour.

I will try adding this as an option to the cleaning code.

kosack, 2016-11-03:
It would be good to implement that as a separate algorithm in `cleaning.py`, so we can compare, rather than adding options to the current algorithm.  The currently implemented algorithm is the "whipple-style" picture/boundary cleaning, so having a separate Hegra-style "tail-cuts" algorithm could be nice. 

The other possibility is to add a second function that modifies the results of the initial cleaning (like the `dilate` function). In whipple this was called "island cleaning", which removed all but the largest island.

kosack, 2016-11-03:
generally, if there are single pixels left, it means the thresholds are set too high.  In the whipple-style algorithm, the intention is not to use, fixed values for the thresholds in PE, but rather fractions of the median value of the pedestal variances for that image (which is why the pedvars are passed in).  The defaults in whipple were picture=5.0% median pedvar, and boundary=2.5% median pedvar.  That does leave some possibility for single pixel fluctuations though,

GernotMaier, 2016-11-03:
Here the link to Isabel's paper on the island cleaning:
http://adsabs.harvard.edu/abs/2003APh....20..311B

commit 0bd51e39 (2017-04-28): fix #248, now have option to do image cleaning where single pixels above pic thresh are removed

- `tailcuts_clean` now has a `keep_isolated_pixels` option, which
defaults to False (but can be turned off in the case it is needed,
since there are some use cases where you want to keep them)
- improved tests of tailcuts_clean
commit dce85375 (2017-04-28): Merge pull request #379 from kosack/master

fix #248, now have option to do image cleaning where single pixels above pic thresh are removed
commit 677a1b51 (2017-05-02): Merge remote-tracking branch 'remotes/upstream/master' into additional_dl1

* remotes/upstream/master: (134 commits)
  added tqdm and iminuit as dependencies
  removed files accidentally committed to Muon Master
  added FAQ and author list
  added author list
  fix size hillas parameter display overlay
  minor bug fixes preparing for merge to master
  fix #248, now have option to do image cleaning where single pixels above pic thresh are removed
  Adding muon fitting tests - still need to make image not a masked array
  add a tool to dump instrument info to FITS files from simtel files
  fix e [... cut]
commit 29bc053a (2018-01-30): Merge pull request #379 from kosack/master

fix #248, now have option to do image cleaning where single pixels above pic thresh are removed
closed 2017-04-28 by kosack