cta-observatory/ctapipe #252: need a way to store telescope patterns/lists in an output table

kosack, 2016-11-15:
Currently a few containers contain a list of telescopes that were e.g.:
* triggered
* read out
* used in reconstruction
* used in classification

These should end up in the output file, but so far the data formats chosen (FITS, HDF5) don't have a good way to store a column with a variable-length array in it (it's very inefficient in FITS, a bit better in HDF5, but very few tools support reading it).  Should we use a more serializable representation?

*  a fixed-length bitfield (each bit representing a telescope), would need to be at least the number of bits as there are telescopes
*  stick with var-length arrays, but take the performance hit?
* other ideas?

MaxNoe, 2016-11-15:
This is a problem for DL1, correct? As DL2 is by definition tabular as far as I understood.

contrera, 2016-11-15:
In principle the fixed-length bitfield looks better suited for me since, even if size is not an important issue because these lists should appear once (or at most a few times) per event. I do not know if this may cause performance issues.

One possibility to explore is if the length of the field can be different in CTA-N and CTA-S.

 Q: ¿ If a fixed length array is used (instead of a bitfield) won´t HDF5 or CFITs compressed it efficiently ?  

  

 

-

maxnoe, 2020-10-26:
This is done now.

closed 2020-10-26 by maxnoe