cta-observatory/ctapipe #254: Get telescope pointing direction

ParsonsRD, 2016-11-18:
Is there a way currently available to get the pointing direction of a given telescope? This is needed for the coordinate transformations, especially for offset or diffuse sources.

kosack, 2016-11-18:
I'm not sure. I'd imagine it's stored somewhere in the simtel files (not just the filename), so a new pyhessio function may be needed to access it. I don't think simtel supports tracking data, so it's not something that changes within a file (just a fixed az/alt).

MaxNoe, 2016-11-18:
It does. There is a Telescope Tracking information container in simtel

ParsonsRD, 2016-11-18:
OK, @watsonjj and I will have a go next week at adding a pyhessio function to get this info.

MaxNoe, 2016-11-18:
In FACT we implemented a pure python reader for the eventio file format. Reading simtel data containers is not supported (yet?), but it might be helpful to see what container are in the files (See the last section of the README).

https://github.com/fact-project/pyeventio

closed 2016-12-12 by ParsonsRD