cta-observatory/ctapipe #257: hessio_event_source broken with pyhessio 1.0

ParsonsRD, 2016-11-22:
When updating the latest version of pyhessio (1.0) a lot of the interfaces for the opening and looping over files. This then breaks lots of the functionality within ctapipe and lots of tests fail. I guess seen as this is a fairly critical dependency we should update the pipeline code also.

jacquemier, 2016-11-22:
This is linked  to PR https://github.com/cta-observatory/pyhessio/pull/39
I was waiting that Karl merge the "large change to container classes"

I will made a PR today for this.
Once accepted, all users will have to upgrade to pyhessio 1.0

If you need an old pyhessio version, you can use this https://github.com/cta-observatory/pyhessio/archive/v0.5.7.zip version of pyhessio

ParsonsRD, 2016-11-22:
Ah great, thanks!

kosack, 2016-11-22:
should be ready to start implementing now (I've merged in most of the new changes)

THis also bring up the fact that we need to make a correct `requirements.txt` for the package (for pip and conda to use during installation) and a conda `yml` file with the correct dependencies for each release for a conda package. It's fine to have a current `ctapipe` release depend on an older `pyhessio`, but it just needs to be explicitly stated in the

closed 2016-11-24 by ParsonsRD