cta-observatory/ctapipe #281: Interim Lookup table format

ParsonsRD, 2016-12-14:
I know in most cases we are trying to steer clear of creating lookup tables, I think it might be worthwhile for the moment to create a simple format just to test the software gives consistent results with current chains for values like MSCW/L. 

I guess ultimately this will be something fits based, but for now it might be easier just to make something numpy array based. For example a pickle file containing a dictionary, where the dictionary key contains the description of the table e.g. [zenith, azimuth] and the corresponding value is the table as a 2D numpy array. Does this sound reasonable, or does anyone have other/better suggestions?

kosack, 2017-01-03:
for really simple tables just use the `ctapipe.core.Histrogram` class. It already does N-dimensional histogramming and FITS input/output.

kosack, 2017-01-03:
Sorry: that should be `ctapipe.utils.Histogram` (the file is in `utils/fitshistogram.py`, but it's imported into the utils namespace)

closed 2017-01-03 by kosack

ParsonsRD, 2017-01-10:
Thanks @kosack I was planning on building a simple class on top of this to perform simple interpolation between lookups for different observing conditions (using scipy interpolate).

kosack, 2017-01-10:
sounds good!