cta-observatory/ctapipe #282: Problems updating pyhessio

AMWMitchell, 2016-12-15:
When trying to follow the update instructions, I encountered the following error:

$ conda install --use-local pyhessio
Using Anaconda Cloud api site https://api.anaconda.org
Fetching package metadata: ..........
Solving package specifications: ....

The following specifications were found to be in conflict:
  - _nb_ext_conf (target=_nb_ext_conf-0.3.0-py35_0.tar.bz2) -> ipywidgets
  - _nb_ext_conf (target=_nb_ext_conf-0.3.0-py35_0.tar.bz2) -> nb_anacondacloud
  - _nb_ext_conf (target=_nb_ext_conf-0.3.0-py35_0.tar.bz2) -> nb_conda_kernels
  - _nb_ext_conf (target=_nb_ext_conf-0.3.0-py35_0.tar.bz2) -> nbpresent
  - _nb_ext_conf (target=_nb_ext_conf-0.3.0-py35_0.tar.bz2) -> python 2.7*|3.4*|3.5*
  - alabaster (target=alabaster-0.7.8-py35_0.tar.bz2) -> python 2.6*|2.7*|3.3*|3.4*|3.5*
  - anaconda-client (target=anaconda-client-1.5.1-py35_0.tar.bz2) -> clyent
  - anaconda-client (target=anaconda-client-1.5.1-py35_0.tar.bz2) -> python 2.7*|3.3*|3.4*|3.5*
  - anaconda-client (target=anaconda-client-1.5.1-py35_0.tar.bz2) -> pyyaml
  - anaconda-client (target=anaconda-client-1.5.1-py35_0.tar.bz2) -> requests *|>=2.9.1
  - anaconda-client (target=anaconda-client-1.5.1-py35_0.tar.bz2) -> setuptools
  - appnope (target=appnope-0.1.0-py35_0.tar.bz2) -> python 2.7*|3.4*|3.5*
  - astropy (target=astropy-1.2.1-np111py35_0.tar.bz2) -> python 2.6*|2.7*|3.3*|3.4*|3.5*
...... (a very long list)

In the end, I had to do a fresh install of conda. Then, the new installation instructions from "Getting Started for developers" did not work:

$ conda install cta-observatory/pyhessio 
Fetching package metadata ...........
Solving package specifications: .
PackageNotFoundError: Package not found: '' Package missing in current osx-64 channels: 
  - cta-observatory/pyhessio

Following the old set of installation instructions then worked.

mdpunch, 2017-01-05:
Noticed this too.  For me it seems to work okay with:
```
cd pyhessio/
git pull 
cd ..
conda build pyhessio
conda install --use-local pyhessio
````
... without any need to re-install conda.

jacquemier, 2017-01-09:
I have just upload pyhessio to anaconda cloud for user cta-observatory [here](https://anaconda.org/cta-observatory/pyhessio)
Please try again with the following command:
conda install -c cta-observatory pyhessio
We will change the documentation

closed 2017-01-09 by jacquemier