cta-observatory/ctapipe #285: Command line argument with multiple args using traitlets

watsonjj, 2016-12-16:
With argparse it is possible to use the "nargs" flag to allow multiple entries for the argument, so it becomes a list. E.g. a list of files (-f ~/Downloads/file1.txt ~/Downloads/file2.txt)

How can this be done for a Tool using traitlets?

watsonjj, 2016-12-19:
I found https://github.com/ipython/traitlets/pull/322

The only way I can make it work is using the List traitlet, and using the cmd line argument:

-f="['/Users/Jason/Software/outputs/lab/2016-12-15/run0.fits', '/Users/Jason/Software/outputs/lab/2016-12-15/run1.fits']"

Obviously this is not very nice, and removes the ability to use ~ and * to specify files.

kosack, 2017-01-03:
Yes, it seems neither argparse or traitlets is perfect! I ran into bad cases with each. 

This however may be possible by defining a custom traitlet for lists of files.  See e.g. http://traitlets.readthedocs.io/en/stable/defining_traits.html   The question is how `traitlets.config.Application` splits up the command-line (it may expect only a single string after each option).  That can be changed though, and we'll probably make a custom version anyhow soon.

An alternative would be to have the file management class do the expansion (using `os.path.expanduser`, `os.path.expandvars`, and `glob.glob` to handle `*` notation), then you would have a "filepattern" option instead of a file list option. It's not quite as flexible though, since then you don't get the benefit of the full shell capabilities.  In reality this isn't a huge issue, since we will launch everything via the grid front-end, which would launch a job per file (and it handles the file parsing), so even just having a single file at a time is not a giant problem, other than for testing purposes or running locally.

watsonjj, 2017-01-03:
Understood. For now I have bypassed this issue through the use of the traitlet config file handling:

c.TFEventFileLooper.file_list=[
        '/Users/Jason/Software/outputs/lab/2016-12-15/run0.fits',
        '/Users/Jason/Software/outputs/lab/2016-12-15/run1.fits',
        '/Users/Jason/Software/outputs/lab/2016-12-15/run2.fits',
    ]

kosack, 2017-05-04:
a hack would be to do the following so the shell converts a glob into a properly formatted list:
```sh
   --files_list="[`ls run*.fits | xargs | sed 's/ /,/g'`]"
```

(note the use of back-tics)

closed 2017-05-04 by kosack